wled/WLED #3950: DMX Proxy Universe bad performance / high latency

Nuet0815, 2024-05-02:
### What happened?

I was trying to setup a simple E1.31 to DMX node with WLED on an ESP32. Therefore I firstly used the [precompiled DMX binaries from srg74](https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/%40Aircoookie), but also compiled the main version [197f47b](https://github.com/Aircoookie/WLED/commit/197f47befe568eef486d006aaec50b06c11c5fc4) myself with dmx enabled.

When I activate the "Proxy Universe 1 from E1.31 to DMX", the data is routed, but with high latency. On ESP32 via Wifi it as constantly around 1s delay between the change I made in my light control software (QLC+) and the DMX device reacting. On ESP32 via Ethernet, it is around 1.5s and on ESP8266 it starts with lower latency but gets worse over time, as if the processor is lagging behind and creates a data jam.

I found a possible, though annoying workaround when I configured the WLED DMX to output RGBW data with no spacing betwen channels via DMX for 128LEDs (128x4 = full 512 channels) and setting the DMX Mode of the Sync Interface to "Multi RGBW". In that way, the DMX is proxied correctly with no latency.

Am I missing some important setting / trick that needs to be done to get it working correctly or is this a bug?

I could reproduce this behaviour on:
-ESP32 via Wifi
-ESP32 via Ethernet (WT32-ETH01)
-ESP8266

### To Reproduce Bug

-Use [precompiled DMX binaries from srg74](https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/%40Aircoookie) and flash on ESP
-Connect RS485 transmitter to ESP IO2 and connect RS485 side to DMX device
-Enable "Proxy Universe 1 from E1.31 to DMX"
-In QLC+ (V 4.13.0) setup E1.31 via unicast to your previously flashed WLED node
-In QLC+ change some channel intensities and object latency between software change and DMX device reacting

### Expected Behavior

Data is proxied with no noticeable latency from E1.31 to DMX.

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.14.3 (build 2404040), WLED 0.15.0-b3 (build 2404120)

### Which microcontroller/board are you seeing the problem on?

ESP8266, ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

BlandSauce, 2024-05-02:
I also ran into this, and I solved it by lowering the MasterTimer value in QLC+ (how you do this depends on your OS). By default, it is constantly outputting at 50hz. Lowering to 40hz worked for me.
I haven't looked into WLED's code yet, but at the time I assumed the delay we see is it working through a buffer of incoming packets that QLC+ is (at default settings) sending quicker than it can process.

blazoncek, 2024-06-23:
Possible duplicate of https://github.com/Aircoookie/WLED/issues/3297
Closing

closed 2024-06-23 by blazoncek

Nuet0815, 2024-06-23:
> I also ran into this, and I solved it by lowering the MasterTimer value in QLC+ (how you do this depends on your OS). By default, it is constantly outputting at 50hz. Lowering to 40hz worked for me. [...]

Thanks for the hint and sorry for no response from my side, I also lowered the QLC+ frequency to 40Hz and it's working like a charm. As DMX-512 has a physically max frequency of ~44Hz this should be quite obvious, so this is no bug of WLED, I simply didn't knew that QLC+ outputs per default 50Hz via ArtNet / E1.31.