wled/WLED #3952: IR Remote don't work with 3 32x8 panels

oltas123, 2024-05-03:
### What happened?

After connecting a 3th panel IR don't work anymore.  If I reduce led number to 512 led's the remote is working. Using 768 led's the remote don't work anymore.

[wled_cfg_WLED-7-RA(2).json](https://github.com/Aircoookie/WLED/files/15196944/wled_cfg_WLED-7-RA.2.json)

### To Reproduce Bug

Configure 3 panels - no working IR. Reduce number of led's to 512 IR Working. Increase Number to 768 - IR don't work.

### Expected Behavior

Don't now

### Install Method

Binary from WLED.me

### What version of WLED?

wled 15.02b, wled 14.3

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-05-03:
Try splitting output into two outputs and report back.

oltas123, 2024-05-04:
Lower the FPS to 22 helps, thanks

closed 2024-05-04 by oltas123