wled/WLED #3955: Leds flickers to full brightness when powering on

Andro-Marian, 2024-05-03:
### What happened?

With the Athom Digital Miocrophone controller when powering on the Master the leds flicker to Full Brightness (And sometimes has wrong color) then going to lowest brightness and starting to increase the brightness to the selected brightness.

https://github.com/Aircoookie/WLED/assets/156662313/97a2163b-a20d-48bb-b278-a5608e6d741c

### To Reproduce Bug

Set the leds to a lower brightness and toggle the Master power from WLed.

### Expected Behavior

The flicker should not occur. The color should be black when turning on and after like 0.5 seconds to start to transition, maybe.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.3

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

I see than if I set the black color before turning it on and selecting a color after the transition is good without any flicker.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-05-03:
All supported LED ICs work as expected. Consider checking [KB](https://kno.wled.ge) for supported LED ICs.
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.
Closing.

closed 2024-05-03 by blazoncek

Andro-Marian, 2025-01-09:
This is not fixed. And it's a bug' On 0.13 works better

DedeHai, 2025-01-09:
does this happen to you on WS2812 as well?

Andro-Marian, 2025-01-09:
Yes. Also without a shifter and on Breadboard with WS2812b

[wled_cfg_Marian-Desk-2.json](https://github.com/user-attachments/files/18362931/wled_cfg_Marian-Desk-2.json)

softhack007, 2025-01-09:
@Andro-Marian it looks like the "white flash" is happening in the moment when the relay powers the strip ?
In this case, it could be that the strip is not in sync with the led datastream at power-on, so the leds "misunderstand" the data and you see a glitch. Unfortunately we cannot do much in software in this case, however @DedeHai has just prepared a PR that might help:
* https://github.com/Aircoookie/WLED/pull/4474

please test.

DedeHai, 2025-01-09:
forgot to mention: we discussed this at length in discord. code is tested.