wled/WLED #3958: Github compile failure for branch  0_15

dosipod, 2024-05-05:
### What happened?

When trying to compile 0_15 branch  on gh , there are errors  , example run here 
https://github.com/dosipod/WLED/actions/runs/8957480973
Compile  for main  is working as expected 

### To Reproduce Bug

https://github.com/dosipod/WLED/actions/runs/8957480973

### Expected Behavior

Compiling 0_15 branch on github should work or needed changes should be highlighted 

### Install Method

Self-Compiled

### What version of WLED?

0_15 branch

### Which microcontroller/board are you seeing the problem on?

ESP8266, ESP32, ESP32-S3, ESP32-S2, ESP32-C3, Other

### Relevant log/trace output

```shell
Error related to missing QuickEspNow.h but might be others as once i tried to add gmag11/QuickEspNow@^0.8.1 it failed with multiple other errors
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2024-05-05:
Hi @dosipod, the quickespnow lib should be included already by this line:
https://github.com/Aircoookie/WLED/blob/3f9a6cae53889898486dae727bbacebc680d6ee0/platformio.ini#L146

I've just re-started the latest gh build for nodemcuv2_160 to verify that there was no library change today that breaks the build - which does happen sometimes. It still looks good for original `0_15`

https://github.com/Aircoookie/WLED/actions/runs/8950518855/job/24601541052

Are you having some extra usermods or other build flags in your repo? If you post the relevant buildenv, I could try if it does work on my local PC.

dosipod, 2024-05-05:
@softhack007 Hey and thanks for looking at this ,  I  am not having any usermod added, i just forked 0_15  to test and add debug builds but did not change anything yet  , https://github.com/dosipod/WLED/blob/0_15/platformio.ini  is not changed   , i kept the repo public  here https://github.com/dosipod/WLED . 
If you fork and enable gh action on main that works but not on 0_15 . 
I will shortly try your linked platformio.ini on gh  as  i am not  having local pio maintained to test without gh

blazoncek, 2024-05-05:
Whatever happens in your fork is your responsibility if upstream works. As far as I can see all CI builds in this repository were successful in main and 0_15 branches.

dosipod, 2024-05-05:
@softhack007  I just copied your linked  platformio.ini and it complied just fine , please merge the same if you could as some of the guys will start to test 0_15 , test run here  https://github.com/dosipod/WLED/actions/runs/8958891098    ,Have a nice day .

@blazoncek There is nothing called my fork yet  as I did not change any thing and hence that is a straight copy from gh  , i seen few changes to gh compile  but sadly i think not a lot  of guys compile and report on gh to test

blazoncek, 2024-05-05:
@dosipod you are missing the point. CI builds in AC repository are successful which means there is nothing to change.

https://github.com/Aircoookie/WLED/actions

dosipod, 2024-05-05:
@blazoncek Not really sure what to do with that info and as said i copied @softhack007  platformio.ini  and now  works

dosipod, 2024-05-05:
I tried again here but this time i imported the repo  and the compile run just fine https://github.com/dosipod/WLED_GH_COMPLIE/actions/runs/8959113704  , must be something wrong with the way i forked  or the beer but will delete that repo and try again , i will close this

dosipod, 2024-05-05:
Forked again and it worked so must be the beer

closed 2024-05-05 by dosipod

blazoncek, 2024-05-05:
You only need to fork once and set up upstream repository.
```
git remote add upstream https://github.com/Aircoookie/WLED.git
```

Then create local branches from remote:
```
git fetch
git switch -c <branch> origin/<branch>
```

From then on:
```
git checkout <branch>
git pull upstream
git push
```

https://stackoverflow.com/questions/1783405/how-do-i-check-out-a-remote-git-branch

dosipod, 2024-05-05:
I think the platformio.ini i copied was from 14 or i messed up with that completely somehow so user error as i see the new fork  has ``` https://github.com/blazoncek/QuickESPNow.git#optional-debug ```  in platformio.ini already .
Sadly i could only do gh compile or codespaces and did not setup cli on the pc or mac