wled/WLED #3963: Unable to save LED preferences when 12mA (WS2815) is selected

moinois, 2024-05-09:
### What happened?

Enabling the automatic brightness limiter and choosing the WS281x and then the 12mA (WS2815) option  in mA/LED nothing happens when pressing Save.

### To Reproduce Bug

1) Click on Config
2) Click on LED preferences
3) Check the Enable automatic brightness limiter
4) Choose WS281x in the drop down below
5) Select 15mA (WS2815) in mA/LED dropdown
6) Click Save

Nothing happens.

If I choose any other LED in the dropdown it is possible to save.

### Expected Behavior

After klicking Save a message should appear saying LED settings saved.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.0-b2, build 2404100

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
The error message I get in the console of my web browser is:
An invalid form control with name='LA0' is not focusable. <input name=​"LA0" type=​"number" min=​"1" max=​"254" oninput=​"UI()​">​
```

### Anything else?

I have the same issue using the app on iOS, but I'm assuming it uses the same web page in the end. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-05-09:
This is fixed in 0_15 branch.

moinois, 2024-05-09:
Okay, I'm currently using the latest beta (beta 2) in 0.15 branch so I assume it's in the pipeline then? I'll update as soon as the next beta is released to verify.

blazoncek, 2024-05-09:
just grab GH CI binary from latest commit.

dosipod, 2024-05-21:
Someone else reported toady it was not saving on discord  and i did try on the very latest as of  commit 4a7ef07 and still the behavior is funky even when selecting  any led type with the limiter unchecked( workaround in video )   , this was reported a while back and I think it has came a long with the introduction of per-output limiter but not 100% sure as I did not have a proper fixture for that to test 
[image]
 Do not kill the messenger :)

h01ger, 2024-06-30:
I can confirm this on 14.4 and the workaround to enable the brightness limiter, then change other options, then save, then disable brightness limiter again, then save again, too.

blazoncek, 2024-07-02:
Please describe *in detail* each step needed to reproduce the problem. Starting with a new un-initialised device.

The video above does not make any sense.

blazoncek, 2024-07-02:
Unable to reproduce. Using 0.15.0-b4

blazoncek, 2024-07-29:
Please post browsers console output.

closed 2024-12-05 by github-actions[bot]