wled/WLED #3967: Can not compile 0_15 with audio reactive user mode for esp32

Intrinsically-Sublime, 2024-05-11:
### What happened?

Using github workflows to compile 0_15 with the audio reactive usermod directly from the Aircookie repo fails with fft errors. Compiling for nodemcu style esp32 module

I am not sure if this is an issue with wled, the audio reactive user mod or the workflows being generated by https://wled-compile.github.io

```
In file included from wled00/usermods_list.cpp:145:0:
wled00/../usermods/audioreactive/audio_reactive.h:199:8: error: 'ArduinoFFT' does not name a type
 static ArduinoFFT<float> FFT = ArduinoFFT<float>( vReal, vImag, samplesFFT, SAMPLE_RATE, true);
        ^
In file included from wled00/usermods_list.cpp:145:0:
wled00/../usermods/audioreactive/audio_reactive.h: In function 'void FFTcode(void*)':
wled00/../usermods/audioreactive/audio_reactive.h:280:7: error: 'FFT' was not declared in this scope
       FFT.dcRemoval();                                            // remove DC offset
       ^
wled00/../usermods/audioreactive/audio_reactive.h:281:22: error: 'FFTWindow' has not been declared
       FFT.windowing( FFTWindow::Flat_top, FFTDirection::Forward); // Weigh data using "Flat Top" function - better amplitude accuracy
                      ^
wled00/../usermods/audioreactive/audio_reactive.h:281:43: error: 'FFTDirection' has not been declared
       FFT.windowing( FFTWindow::Flat_top, FFTDirection::Forward); // Weigh data using "Flat Top" function - better amplitude accuracy
                                           ^
wled00/../usermods/audioreactive/audio_reactive.h:283:20: error: 'FFTDirection' has not been declared
       FFT.compute( FFTDirection::Forward );                       // Compute FFT
                    ^
*** [.pio/build/custom_build/src/usermods_list.cpp.o] Error 1
========================== [FAILED] Took 5.41 seconds ==========================

Environment    Status    Duration
-------------  --------  ------------
custom_build   FAILED    00:00:05.408
==================== 1 failed, 0 succeeded in 00:00:05.408 ====================
Error: Process completed with exit code 1.
```

### To Reproduce Bug

This is my github workflow (based on https://wled-compile.github.io )

```
name: WLED esp32-SR-15-nightly compile

on:
    workflow_dispatch:

jobs:
  build:

    runs-on: ubuntu-latest
    steps:
    - uses: actions/checkout@v4
    - name: Set up Python
      uses: actions/setup-python@v5
      with:
        python-version: 3.9

    - name: Install PlatformIO
      run: |
        python -m pip install --upgrade pip
        pip install --upgrade platformio
    - name: Set up Node.js
      uses: actions/setup-node@v4
    - name: Run PlatformIO
      run: |
        rm -rf /tmp/WLEDtemp/* || true
        mkdir -p /tmp/WLEDtemp
        wget https://github.com/Aircoookie/WLED/archive/refs/heads/0_15.zip -O /tmp/WLEDtemp/mm.zip
        unzip /tmp/WLEDtemp/mm.zip -d /tmp/WLEDtemp
        find /tmp/WLEDtemp -maxdepth 1 -mindepth 1 -name "*" -type d >> /tmp/WLEDtemp/wledfoldername
        cd `cat /tmp/WLEDtemp/wledfoldername`
        npm install
        echo '[env:custom_build]' >>platformio.ini
        echo 'extends = env:esp32dev' >>platformio.ini
        echo 'build_flags = ${env:esp32dev.build_flags} -D WLED_ENABLE_PIXART -D USERMOD_AUDIOREACTIVE' >>platformio.ini
        echo 'lib_deps = ' >>platformio.ini
        echo '  ${env:esp32dev.lib_deps}' >>platformio.ini
        echo '  https://github.com/blazoncek/arduinoFFT.git' >>platformio.ini
        
        pio run -e custom_build || pio run -e custom_build
    - name: Copy Files
      run: |
        cd $GITHUB_WORKSPACE
        git rm -r $GITHUB_WORKSPACE/esp32-SR-15-nightly.bin || true
        WLEDFOLDERNAME=`cat /tmp/WLEDtemp/wledfoldername`
        cp ${WLEDFOLDERNAME}/build_output/firmware/custom_build.bin $GITHUB_WORKSPACE/
        date -I | tr -d '\n' >$GITHUB_WORKSPACE/last_update_date.txt
    - name: Commit Files
      run: |
        cd $GITHUB_WORKSPACE
        git config user.name "GitHub Actions Bot"
        git config user.email "<>"
        git add $GITHUB_WORKSPACE/esp32-SR-15-nightly.bin
        git add $GITHUB_WORKSPACE/esp32-SR-15-nightly.last_update_date.txt
        git commit -m "Commit Custom Build esp32-SR-15-nightly"
        git push origin main
```

### Expected Behavior

For the compile to complete.

### Install Method

Self-Compiled

### What version of WLED?

0_15

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2024-05-11:
Hi,

Please use [Discord](https://discord.gg/QAh7wJHrRM) or WLED [forum](https://wled.discourse.group/) for help and support questions. Github CI builds in the Aircoookie repo work flawlessly, so it seems you have a problem in your own github scripts. Closing.

<br/>

It looks like your on-the-fly build script is having an outdated version of ArduinoFFT that's no longer supported In 0_15.

https://github.com/Aircoookie/WLED/blob/c016dedea4c74d35d0ce36c35e1b6e96eb711932/platformio.ini#L228-L229

https://github.com/Aircoookie/WLED/blob/c016dedea4c74d35d0ce36c35e1b6e96eb711932/platformio.ini#L374-L380

closed 2024-05-11 by softhack007

Intrinsically-Sublime, 2024-05-11:
@softhack007 Thank you for the answer. Everywhere I had read said to use blazoncek/arduinoFFT.git for audio reactive. I will see if I can make a pull request for the compile helper script.

As for Discord and Discourse I will never use them. I can participate and provide stuff like the ledmapping software I wrote recently and report bugs etc. But if I am forced to use a forum or social media I will just keep my work to myself and hope the project in question fails. Fracturing a project across as many possible sites is simply terrible. You could enable discussion here on github and keep the discussions in one place. But no it has to be spread everywhere including black holes for information that are not indexed by search engines. Then everyone gets upset when the same questions get asked over and over because no one can find them spread across the internet.

blazoncek, 2024-05-11:
> Everywhere I had read said to use blazoncek/arduinoFFT.git for audio reactive.

That was valid until 2.0 of ArduinoFFT was released.

softhack007, 2024-05-11:
... and the [updated  audioreactive readme](https://github.com/Aircoookie/WLED/tree/0_15/usermods/audioreactive#installation) in our 0_15 branch says the same. It means instead of asking google, you might get better results by reading original docs.

It may be a bit frustrating, but sometimes the internet, YouTube or even chatGPT have very old information, and a good old round of _RTFM_ provides best results.

softhack007, 2024-05-11:
Update and good news for all friends of custom self-compile scripts:

A) all latest head branches (AC WLED/main, AC WLED/0_15, MoonModules WLED/mdev) should now build succesfully with 
* build_flags = `-D USERMOD_AUDIOREACTIVE -D UM_AUDIOREACTIVE_USE_NEW_FFT`
* lib_deps = `kosme/arduinoFFT @ 2.0.1`

B) for older versions, you will succeed with 
* build_flags = `-D USERMOD_AUDIOREACTIVE` 
* lib_deps = `https://github.com/blazoncek/arduinoFFT.git`

C) The other option that works since 0.14.1 (and MM 0.14.1-b32) is 
* build_flags = `${esp32.AR_build_flags}` 
* lib_deps = `${esp32.AR_lib_deps}`
this will give you recommended settings for audioreactive automatically.

Cc @wled-install @lost-hope @dosipod This info might be useful for your build bots 😃 

To be clear: this is an "Extrawurst" (something special) from my side, not a guaranteed and 100% supported feature.

dosipod, 2024-05-13:
@softhack007 Thank you yes i did notice the error few days back when we started testing 15 compile on gh and changed it  to kosme/arduinoFFT @ 2.0.1  , I was about to get in touch with @wled-install ( if that is wladi ) on discord to point that out if he would change it in https://wled-install.github.io/ ( if he did not do that already which i am yet to test ) .
Really appreciate the working you are doing to simplify compiling , please keep it up