wled/WLED #3969: WS2805

mathojojo, 2024-05-12:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Thank you for your ideas for making WLED better!

mathojojo, 2024-05-12:
Sorry, There is already an existing issue for WS2805. i close  it

closed 2024-05-12 by mathojojo

blazoncek, 2024-05-13:
It is implemented.

mathojojo, 2024-05-13:
Thank you @blazoncek. Can you please tell me which version I should get to test it (probably not the official realease) ?