wled/WLED #3970: Could not copmpile

larryl79, 2024-05-13:
### What happened?

Tried to compile source to binary for ESP8266.

### To Reproduce Bug

latest source with usermod rotary and 4 line oled LCD
latest VSC and PIO
using this methoed to compile: https://www.instructables.com/ESP8266-WLED-Hardware-Experimentation-Platform/

### Expected Behavior

Compile without issue.

### Install Method

Binary from WLED.me

### What version of WLED?

latest source

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell
VSC report:

	"resource": "/d:/Documents/PlatformIO/Projects/Wled with usermod/wled00/FX.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "ambiguous overload for 'operator==' (operand types are 'CRGB' and 'CRGB::HTMLColorCode')",
	"startLineNumber": 5227,
	"startColumn": 24,
	"endLineNumber": 5227,
	"endColumn": 24
	
	
	"resource": "/d:/Documents/PlatformIO/Projects/Wled with usermod/wled00/FX.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "ambiguous overload for 'operator==' (operand types are 'CRGB' and 'CRGB::HTMLColorCode')",
	"startLineNumber": 5238,
	"startColumn": 24,
	"endLineNumber": 5238,
	"endColumn": 24
```

### Anything else?

changed source on my computer:

5227 -- if ((oldSpawnColor == CRGB::Black) || (oldSpawnColor == trailColor)) oldSpawnColor = spawnColor; // reject "black", as it would mean that ALL pixels create trails
5227 ++ if ((oldSpawnColor == (CRGB)CRGB::Black) || (oldSpawnColor == trailColor)) oldSpawnColor = spawnColor; // reject "black", as it would mean that ALL pixels create trails

5238 -- if (pix != CRGB::Black) SEGMENT.setPixelColorXY(col, row, pix.nscale8(fade)); // optimization: don't fade black pixels
5238 ++ if (pix != (CRGB)CRGB::Black) SEGMENT.setPixelColorXY(col, row, pix.nscale8(fade)); // optimization: don't fade black pixels

no compile issue anymore

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

larryl79, 2024-05-13:
pull request created.

blazoncek, 2024-05-13:
Please use WLED [forum ](https://wled.discourse.group)or Discord for help and support questions. You can also consult WLED [KB](https://kno.wled.ge/advanced/compiling-wled/).
Using correct set up, everything compiles successfully.

closed 2024-05-13 by blazoncek