wled/WLED #3972: Anymore then two output pins causing glitching (ESP32-WROOM-32)

1liminal1, 2024-05-13:
### What happened?

I have five pins with 768 LEDs each (4,16,17,18,19). Under LED preferences when adding any more then two outputs, it starts to glitch. If I add any two pins that are at a

https://github.com/Aircoookie/WLED/assets/10505942/0af09023-61fb-482a-ad8a-3a51f4a3850b

ny location they are fine, but when adding more then two they start glitching and get worse with more pins added. 

### To Reproduce Bug

Add any more then two pins on a ESP32-WROOM-32 and the LED with start to glitch like there is noise on the PIN. The more hardware PINS added the more it glitches with the first two running smoothly, subsequent additions (in any physical hardware order) will glitch progressively worse  

### Expected Behavior

No glitching 

According to WLED https://kno.wled.ge/features/multi-strip/ adding more PINs should distribute the load? and be better for refresh rates. 

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0-b2

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
Please advise what you need?
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

1liminal1, 2024-05-13:
If I do two outputs @ 1536, they will be quite stable, but the third @ 768px will glitch like mad.

blazoncek, 2024-05-13:
Try experimental build of 0_15 branch and use WLED_USE_PARALLEL_I2S. Be mindful that each output cannot exceed 256 LEDs due to memory issues.
You need to compile yourself.

1liminal1, 2024-05-13:
Thank you, I appreciate it, but this documentation says it should work? 

This recommends https://kno.wled.ge/features/multi-strip/  that keeping each channel (what are we calling it?) under 800px and that should be fine for high speeds. Or am I missing something?

Is this a bug, or the hardware is not powerful enough?

I need to drive a total of 3840px

softhack007, 2024-05-13:
In principle 8 outputs do work. With the number of LEDs you want to drive I would expect that an esp32 can achieve between 10 and 30 fps (frames per second). However the glitches you describe are something we probably cannot influence in the WLED core software.

* electrical issues: make sure to follow the [recommended electrical design](https://kno.wled.ge/basics/getting-started/), including resistor between esp32 pin and LEDs dataIn.  Make sure to solder all data connections (I don't know what below your shrink tubes..)
* issues in the led driver: we use [NeoPixelBus as driver library](https://github.com/Makuna/NeoPixelBus). The timing of RMT outputs is very critical, and when the driver does not react to interrupts in time, there can be intermittent glitches. Using more outputs increases the "pressure" on interrupt handling in NPB and may lead to glitches. To verify that the timing of outputs is correct, use a logic analyzer. In case of issues, please file a bug report for NeoPixelBus.

dDriver timing issues can sometimes be reduces by using a higher flash speed, however not all esp32 work with such "overclocking". To try out, add these two lines in your custom build environment, then build and upload from the VSCode+platformIO development environment.
```ini
board_build.f_flash = 80000000L
board_build.flash_mode = dio ;; or dout
```

1liminal1, 2024-05-14:
Thank you so much for the info. I appreciate it :) 

Question about electrical. SO I'm using 5v all the way though, psu though to the 5v LED panels and then off to the 5v on the ESP32. SO, not sure a level shifter is relevant (I also understand that you don't have a lot of context for my set up, so just giving me generic info :)

My total LEDS are 3840 and i'm not using a resistor, happy to try that. But the issue I'm having is the more outputs I add, the more glitching. 

If I run two outputs, one @ 1536PX and the other @ the max of 2048, the second output gets very little glitching. Even though I get all these warning in WLED that that I should add more pins (don't quote me, I don't have access to it now). If I run the full 5 outputs all @ 768px each. The 3rd, 4th and 5th glitch with increasing intensity. SO the 3rd less building though to the 5th more. 

If I change the Outputs around on different pins, again, the first two is ok and as I add more, starting at the third Output, it starts to glitch. This seems like a bug to me or a sign that the documentation is wrong is relation to 2d matrices?

I will try the overclock and see if that works or helps. But if you're saying this is not a bug, then it appears that, at least with 2d matrices, running less outputs with more pixels is what works and not what the documentation suggests.

1liminal1, 2024-05-14:
Just to add; 

All connections are soldered. with JST connectors with the power removed for addition data connectors 

[image]

blazoncek, 2024-05-14:
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.

As stated above build your own custom version with experimental parallel I2S support. If that does not work, use 0.14.3 with 4 outputs as more than 5 will hit the bug in RMT driver code.

1liminal1, 2024-05-14:
Just to help anyone that comes across this. I added a 62ohms resistor to the data line, close to the LEDs and it made absolutely no difference to the glitching, it may help someone, but probably don't waste the time.

robertvb83, 2024-05-15:
I see several problems 

You have only one main power supply with thin wires. With that many LEDs you have immensely high Current demand.

And then you have only 1 Power injection per 3 panels, this is way too less. Add power injection for each panel.

What size is your PSU? It must be huge!

And the next issue is that you are not using a level shifter, the ESP32 output is not according to the LED panel specification. 5V as input power to ESP32 does not mean that GPIO output is 5V. The level shifter makes sure that these signals are properly adjusted to the LEDs spec.

Usually for very short data lines things work well without LS, longer Data lines are problematic and lead to flicker.

I am building a Matrix with 8 panels and 4 LED outputs. That works without overclocking, without parallel I2S with latest branch 0.15 code

1liminal1, 2024-05-15:
Thanks for that, but the power injection runs up the middle where the XT60s are :) 

Its 20a, but we are not looking for brightness here, it's a feature light, so you can get away a lower brightness. You just want it to be consistent :) 

I don't power the LEDs from the ESP32, its the other way round, the LEDs power the ESP32. 

[image]

Power goes up the middle though the power injection. then I have Vin and GND off of one of the panels to power the ESP32, which also gives the first Output of LEDs 

You can see it here, along with the other data Outputs 

[image]

closed 2024-09-20 by github-actions[bot]