wled/WLED #3973: Can't compile 0_15 - No option 'build_flags_esp8266' in section: 'common'

DarthWeber, 2024-05-13:
### What happened?

After updating with github i can't compile anymore:

`InvalidProjectConfError: Invalid 'C:\Users\tomwe\Dokumente\GitHub\WLED15\platformio.ini' (project configuration file): 'No option 'build_flags_esp8266' in section: 'common''`

### To Reproduce Bug

compile latest release of 0_15 branch

### Expected Behavior

compile...

### Install Method

Self-Compiled

### What version of WLED?

0_15_b3

### Which microcontroller/board are you seeing the problem on?

Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-05-13:
Your fork or clone is likely corrupt. Please re-pull everything.
If you use `platformio_override.ini` you will need to update it to be compatible with definitions in `platformio.ini`.

closed 2024-05-13 by blazoncek

softhack007, 2024-05-13:
in 0_15 you need to use `${esp8266.build_flags}`