wled/WLED #3975: Persist power state on reboot

drewcovi, 2024-05-13:
**Is your feature request related to a problem? Please describe.**
Seems with some recent updates, I'm now facing (and may have been facing for some time) unexpected restarts. This means installations in my kitchen for instance triggers a preset in the middle of the night when it was turned off before bed.

**Describe the solution you'd like**
It's not uncommon to see a feature in smart home lighting that restarts with whatever power state they were in prior to losing power or power cycling.

**Describe alternatives you've considered**
Autosave usermod does allow for storing any preset, and perhaps the "off" preset could be considered one. 

**Additional context**
Goes without saying that this might be a usability improvement as WLED appeals to a broader audience. Usermods require custom compilation, so this would go a long way to rendering power cycle issues moot and aligning even closer to common smart lighting conventions.

Thank you for your ideas for making WLED better!

Thank you for making WLED 😉

blazoncek, 2024-05-13:
Saving too often wears flash down and met render device inoperable over time.
It will not be implemented as a core feature unless Autosave usermod is incorporated into default builds.

drewcovi, 2024-05-13:
Really great point. Does Autosave also wear out the flash? I guess I'll just cross my fingers on the power cycle issue. So long as uptime is strong, it's probably a non-issue (only during rare power outages).

blazoncek, 2024-05-14:
Any writ does. My own (limited) testing showed that even as low as 40000 writes can destroy flash. And there are a lot of writes that happen.

encrust-ace, 2025-06-05:
I’m just curious about how platforms like WiZ manage this so well with longterm stability.

For example, let’s say I’m away on vacation and there’s a power cut. When power is restored, all the lights turn on.
Even if I had previously disabled the “power-on after power restore” option and there’s a power outage during a party, I’d have to manually turn everything back on again.

I know this can be handled with groups and automations in Home Assistant, but I’m genuinely curious how does WiZ handle this kind of situation so reliably?

blazoncek, 2025-06-05:
> I’m just curious about how platforms like WiZ manage this so well with longterm stability.

You will need to ask Wiz about that.