wled/WLED #3976: 0.15.0-b2 causes flashing. Downgrade to 0.14.3 resolves the issue with no other changes

nomonkeynodeal, 2024-05-13:
### What happened?

Upgrading a Dig-Quad (ethernet) from 0.14.3 to 0.15.0-b2 results in weird flashing. See video: https://imgur.com/a/CXnvmKY

The flashing only seems to occur on GPIO 16 and 1. It does not occur on GPIO 3. Additionally, I am not seeing the flashing with the solid effect. I can see it with a several effects, and it's really visible with Colortwinkles in the video.

After rolling back to 0.14.3 the flashing no longer occurs.

I have three other Dig-Quads (ethernet as well) and a Dig-Octa (ethernet) that upgraded without issue. I am mystified as to what the difference can be. They are all running the same effect (Candle Multi) usually.

### To Reproduce Bug

.

### Expected Behavior

.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.0-b2 (build 2404100)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-06-23:
Please use latest available code (build yourself if necessary).

nomonkeynodeal, 2024-06-23:
> Please use latest available code (build yourself if necessary).

B3 does the same thing.

blazoncek, 2024-06-23:
You will need to provide more detail then.
Including output from debug build.

rafurius, 2024-10-08:
Me ocurre lo mismo, ESP32,  ver: 0.15.0-b5  GPIO 14.

hacustom, 2024-12-13:
Thank you for posting this, I thought I was going crazy after upgrading to the full release of 0.15.0 and my lights completely **** the bed.  I'm also using a DigQuad, I have 3 segments 2 of which are on GPIO 1 and 16.  My segment on GPIO 3 was fine but the other two were just random color noise.  I thought something ate through wiring or something but as soon as I downgraded back to 0.14.3 all was well.

ciaranj, 2024-12-17:
I think I'm seeing this too, ESP32, a segment on GPIO 2 flashes occasionally, but segments on GPIO 4, 18 & 19 don't seem to. No issues prior to the 15.0 upgrade, don't think I saw the issue on b3 though either! It may also be rebooting at the time it flashes, as the uptime is much lower than I expect.

hacustom, 2025-04-17:
To the best of my knowledge this is still an open defect is it not?

blazoncek, 2025-07-12:
Related to #4389 
Closing as a duplicate.

closed 2025-07-12 by blazoncek