wled/WLED #3981: FW1906 single white flash when turned on

Suxsem, 2024-05-15:
### What happened?

I have a long strip of FW1906 leds (RGBCCT) and a relay to turn off the power supply when not in use (using the built-in relay feature of WLED). WLED controller is always powered.
Every time I turn on the strip I get a brief (like 0.1s or so) white flash at full brightness before getting the desired color/brightness.
Any help?
Thank you!

### To Reproduce Bug

.

### Expected Behavior

.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.0-b3

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Suxsem, 2024-05-26:
No one experienced this?

softhack007, 2024-05-26:
In case you use an SSR relay to control AC power - it might be  a hardware problem.
* https://github.com/Aircoookie/WLED/issues/3581

Another possibility is is you're using TX or one of the esp32 "strapping pins". These are driven by the bootloader, when WLED is not running yet. In this case, try a different pin for LEDs driving.

closed 2024-10-01 by github-actions[bot]