wled/WLED #3982: WLED compile helper not working

Danit2, 2024-05-15:
### What happened?

I have try the manual from the WLED compile helper site to compile my one WLED bin. 

But it is not working. 

I become this error:
```
Run cd $GITHUB_WORKSPACE
  cd $GITHUB_WORKSPACE
  git rm -r $GITHUB_WORKSPACE/*.bin || true
  WLEDFOLDERNAME=`cat /tmp/WLEDtemp/wledfoldername`
  cp ${WLEDFOLDERNAME}/build_output/firmware/custom_build.bin $GITHUB_WORKSPACE/
  date -I | tr -d '\n' >$GITHUB_WORKSPACE/last_update_date.txt
  shell: /usr/bin/bash -e {0}
  env:
    pythonLocation: /opt/hostedtoolcache/Python/3.9.19/x64
    PKG_CONFIG_PATH: /opt/hostedtoolcache/Python/3.9.19/x64/lib/pkgconfig
    Python_ROOT_DIR: /opt/hostedtoolcache/Python/3.9.19/x64
    Python[2](https://github.com/Danit2/wled_battery/actions/runs/9094335217/job/24995178211#step:7:2)_ROOT_DIR: /opt/hostedtoolcache/Python/[3](https://github.com/Danit2/wled_battery/actions/runs/9094335217/job/24995178211#step:7:3).9.19/x64
    Python3_ROOT_DIR: /opt/hostedtoolcache/Python/3.9.19/x6[4](https://github.com/Danit2/wled_battery/actions/runs/9094335217/job/24995178211#step:7:4)
    LD_LIBRARY_PATH: /opt/hostedtoolcache/Python/3.9.19/x64/lib
fatal: pathspec '/home/runner/work/wled_battery/wled_battery/*.bin' did not match any files
cp: cannot stat '/tmp/WLEDtemp/Aircoookie-WLED-3043[5](https://github.com/Danit2/wled_battery/actions/runs/9094335217/job/24995178211#step:7:5)e6/build_output/firmware/custom_build.bin': No such file or directory
Error: Process completed with exit code 1.

```

### To Reproduce Bug

Folow the step by step manual on the WLED compile helper site. 

### Expected Behavior

That the manual will work. 

### Install Method

Self-Compiled

### What version of WLED?

v0.15.0-b2

### Which microcontroller/board are you seeing the problem on?

ESP32-C3

### Anything else?

Here the compile yaml:
```
name: WLED compile

on:
    workflow_dispatch:

jobs:
  build:

    runs-on: ubuntu-latest
    steps:
    - uses: actions/checkout@v4
    - name: Set up Python
      uses: actions/setup-python@v5
      with:
        python-version: 3.9

    - name: Install PlatformIO
      run: |
        python -m pip install --upgrade pip
        pip install --upgrade platformio
    - name: Set up Node.js
      uses: actions/setup-node@v4
    - name: Run PlatformIO
      run: |
        rm -rf /tmp/WLEDtemp/* || true
        mkdir -p /tmp/WLEDtemp
        wget https://api.github.com/repos/Aircoookie/WLED/zipball/v0.15.0-b2 -O /tmp/WLEDtemp/mm.zip
        unzip /tmp/WLEDtemp/mm.zip -d /tmp/WLEDtemp
        find /tmp/WLEDtemp -maxdepth 1 -mindepth 1 -name "*" -type d >> /tmp/WLEDtemp/wledfoldername
        cd `cat /tmp/WLEDtemp/wledfoldername`
        npm install
        echo '[env:custom_build]' >>platformio.ini
        echo 'extends = env:esp32c3dev' >>platformio.ini
        echo 'build_flags = ${env:esp32c3dev.build_flags} -D USERMOD_BATTERY' >>platformio.ini
        
        pio run -e custom_build || pio run -e custom_build
    - name: Copy Files
      run: |
        cd $GITHUB_WORKSPACE
        git rm -r $GITHUB_WORKSPACE/*.bin || true
        WLEDFOLDERNAME=`cat /tmp/WLEDtemp/wledfoldername`
        cp ${WLEDFOLDERNAME}/build_output/firmware/custom_build.bin $GITHUB_WORKSPACE/
        date -I | tr -d '\n' >$GITHUB_WORKSPACE/last_update_date.txt
    - name: Commit Files
      run: |
        cd $GITHUB_WORKSPACE
        git config user.name "GitHub Actions Bot"
        git config user.email "<>"
        git add $GITHUB_WORKSPACE/custom_build.bin
        git add $GITHUB_WORKSPACE/last_update_date.txt
        git commit -m "Commit Custom Build"
        git push origin main

```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2024-05-16:
Regardless of the compile helper  error which is see also , might be @softhack007  could advice as we have seen issues with c3 even in ardunio when trying to use analogRead , not really sure that is specific to some c3 boards on in general as i only have one

softhack007, 2024-05-16:
Hi @dosipod and @Danit2, well this "compile helper" is not any official feature, so you should first contact the original author of the guide.

That said, i took a quick look at your build job

https://github.com/Danit2/wled_battery/actions/runs/9094335217/job/24995178211

```
bin_rename_copy([".pio/build/custom_build/firmware.bin"], [".pio/build/custom_build/firmware.elf"]) 
Copying .pio/build/custom_build/firmware.bin to build_output/release/WLED_0.15.0-b2_ESP32-C3.bin 
======================== [SUCCESS] Took 120.28 seconds ======================== 
```

So the build itself is successfull. It looks like the script then tries to copy the firmware.bin from the wrong path. That's a problem that the author of the scripts needs to solve.

There was a change in 0_15 (https://github.com/Aircoookie/WLED/commit/7ee4b54154cdb21e4d066fd22af8a02b553898f3) maybe two months ago about where files are placed in the `build_output` folder.
* builds with -D WLED_RELEASE_NAME=... go into release/
* builds without the flag go into firmware/

This change only affects the 0_15 branch.

But again, we're talking about a compile helper thingy created by some wled user, its _not_ any official and supported feature.

A possible solution for the script could be add this into the temporary buildenv: `build_unflags=  -D WLED_RELEASE_NAME`

Danit2, 2024-05-16:
Thanks @softhack007 for your help.

It wasn't entirely clear to me that this wasn't from WLED itself. Sorry for that.

Nevertheless, thank you very much for your help.

Maybe @wladwnt can help here?

wled-compile1, 2024-05-16:
@Danit2 is updated with a solution as proposed by softhack007 (Thank you!), works now again.

Danit2, 2024-05-17:
Thanks to all for the quick help.

Now it is working again.

closed 2024-05-17 by Danit2