wled/WLED #3983: Config resets after exiting out of the app controlling WLED

ArySlayer, 2024-05-15:
### What happened?

"WLED is receiving live USB Adalight/TPM" data" The Configuration applies successfully but after exiting of Prismatik for example. The configuration returns to the default configuration. Turning this option off changes nothing: "Turn LEDs on after power up/reset:"

### To Reproduce Bug

1. Set preset
2. Exit out of prismatik
3. preset has been reset to default config

### Expected Behavior

You set your preset, with led preferences and save the configuration, after exiting Prismatik the LEDs should simply turn off, and when turned on again the saved config should apply

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.1

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Anything else?

Similar issue: https://github.com/Aircoookie/WLED/issues/1482

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2024-05-16:
Somewhat  hard to replicate due to the following :
A) What is meant by "The configuration returns to the default configuration" and "preset has been reset to default config"  does that mean wled goes to factory reset ? 
B) Have you tested with Hyperon also to rule out issues with Prismatik and which versions tested  ?
C) Why not try with 15 
If you need to confirm from our side  ( might be you do not need that ) then please test and clarify the above

blazoncek, 2024-05-16:
Live data is always only temporary. If the *configuration* (i.e. settings) is changed it needs to be sent via valid JSON API.

ArySlayer, 2024-05-16:
A) WLED was factory reset
B) Hyperion has the same problem
C) Didn't work
I also tried using older versions which had the same issue

blazoncek, 2024-05-16:
So it is not a bug, but a feature.

ArySlayer, 2024-05-16:
> So it is not a bug, but a feature.

So how do disable this feature

blazoncek, 2024-05-16:
By writing custom WLED. 🤷‍♂️

mrdude2478, 2024-06-10:
@ArySlayer 

I had the same issues with configs not saving, this was down to my boards partitions not being able to save the config. Once I created a new partitons bin file and flashed that I was able to save config files. I had a 4MB flash and so I made my partitons file like this:

# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x160000,
app1,     app,  ota_1,   0x170000,0x160000,
spiffs,   data, spiffs,  0x2D0000,0x120000,
coredump, data, coredump,0x3F0000,0x10000,

That gave enough room to fit wled firmware + ota and enough room in nvs and spiffs to save config stuff. After that I didn't get any issues with saving configs.

closed 2024-10-28 by github-actions[bot]