wled/WLED #3985: Bin File for 10 GPIO Buttons

sci666-2023, 2024-05-16:
hello there, i dont know what happens it seems that my ESP8266 are a bit strange because after Version 0.13 the WLAN is so unstable that its not usable. i tried to compile a version 14 with 10 buttons and it works so far but wlan is still not usable .... maybe i do something wrong .... can anybody compile an actual version with 10 GPIO Button support for me or alternatively a Version 13 with 10 Button support ?

also standard 14 and 15 does not work well :/

i got these ones from amazon:
https://www.amazon.de/AZDelivery-NodeMCU-ESP8266-Development-unverl%C3%B6tet/dp/B07Z69WT62/ref=sr_1_3_pp?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=1OQI0HCJA60GQ&dib=eyJ2IjoiMSJ9.af-KzXtYzgCUqLLSR5Q9-LbBIfNvrd46X7XzbXjAatNaE03lWbntXAyRW8yeXkS0eECAqEae6jtLn2a46_1VcC3EdgbWUH8IPPmfpHJp2dl6kVQEP-SrJUSsYp_IPWgWl1G2yQFIEuFfB0T0isLxzNjNlVx-sXP-JckBAyuX4v5OxVCMuhIDirvd9EPe5tlsV5uqIpfWQmVGXOERHhIcTkfa37w_xfaoXusr82NFyDM.bmczCIsD0-SSKxmFSGlGV7SLWFi9jViKnWCoaF3hw4M&dib_tag=se&keywords=esp8266&qid=1715859811&sprefix=esp8266%2Caps%2C102&sr=8-3

softhack007, 2024-05-19:
If you are not sure about your custom firmware, it's best to go to https://install.wled.me/ and install your device freshly with an official firmware. 

Btw, this issuer tracker is meant for bug reports and for proposing new feature. It's not meant for asking help getting your device to work. On our WLED discord community you'll find lot of experienced user who will be happy to help.

Please use WLED forum or [Discord](https://discord.gg/QAh7wJHrRM) for help and support questions. You can also consult WLED [KB](https://kno.wled.ge/advanced/compiling-wled/).

softhack007, 2024-05-26:
Closing, as this is not a software bug.

closed 2024-05-26 by softhack007