wled/WLED #3987: glitter fx freezed gradient

Scarala, 2024-05-17:
### What happened?

I noticed a bug in the glitter fx. The glittering works fine but the color gradient effect is freezed and not a moving gradient like in the older versions of WLED. It seems like the gradient is showing only the 1st frame of the animation but does not start the animation. 
According to the Discord Beta-Testing Channel blazoncek noticed to me, that Glitter was changed on Jan 6th 2023 and sooms to be buggy since then.

### To Reproduce Bug

Just use the Glitter FX

### Expected Behavior

Glitter with a running gradient in background

### Install Method

Binary from WLED.me

### What version of WLED?

15.0-b2

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-05-18:
FYI caused by #3683 by @TripleWhy. The problem lies in the fact that default behaviour of Palette effect changed and cannot be modified from Glitter. Glitter uses Palette effect as a base.

I'll modify Glitter to use old Palette code instead of calling Palette effect. That should resolve the issue.

Scarala, 2024-05-21:
Confirm fixed in 0.15.0-b3

closed 2024-10-27 by blazoncek