wled/WLED #3988: ESP32 unable to update from V0.14-> V0.15

lodesmets, 2024-05-18:
### What happened?

I have an ESP32-D0WDQ6, and usually the OTA works just fine, but I can't seem to update to v0.15.x.

Usually I build custom FW with u usermod active (USERMOD_SSDR), at first I thought it had to do with my custom FW, but it seems that also the "official" esp32 isn't working 

I tried manually gzipping the file, but that fails immediately (in issue https://github.com/Aircoookie/WLED/issues/3915 you say this works for ESP8266 devices 

### To Reproduce Bug

Update a V0.14 ESP32-D0WDQ6 device to V0.15

### Expected Behavior

Update successful

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0 - b3

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-05-18:
You will need to flash it serially as your partition map is too old and the new binary no longer fits.
Be sure to backup configuration and presets as they will be lost otherwise.

lodesmets, 2024-05-19:
Hey, 
It indeed is an older installation. But does this mean I always need to do an USB update in the future? or only this time? 

This might be a dumb question: isn't it possible to update to a version that changes the partition map? Because I'm not going to be the only one with this issue?

softhack007, 2024-05-19:
Did a quick research - unfortunately its not possible to update the partitions table by OTA (see also this [discussion from tasmota](https://github.com/arendst/Tasmota/discussions/12246#discussioncomment-806817) ).

I think tasmota has a special app called "partition manager", however that one will install the tasmota partitioning which is  different from the WLED partitions. It would be a project of it's own to make such a thing for WLED. Actually I don't see how to do that with the availeable development team - unless someone volunteers.

The 1MB littlefs partitioning exists since 0.13. Actually with 0.15 we are again at 98% flash usage for program - so I'd expect that we will need a new partitioning file soon (maybe 500kb littlefs, so we have around 1.7MB program space).

blazoncek, 2024-05-19:
> But does this mean I always need to do an USB update in the future? or only this time? 

Depends on the growth of WLED binary and the actual partition map.

> This might be a dumb question: isn't it possible to update to a version that changes the partition map? Because I'm not going to be the only one with this issue?

No, as the OTA library isn't capable of doing that.

lodesmets, 2024-05-31:
Thanks for the info. I won't close the issue, so Aircoookie sees it maybe, and adds a warning to the documentation/ release note

softhack007, 2024-05-31:
> , so Aircoookie sees it maybe, and adds a warning to the documentation/ release note

@lodesmets the warning is already there, in the release notes for 0.13.3

[image]

The only news is that -since 0.14.0 - the old partition layout does not allow OTA update, as the new firmware is too big for the old OTA partition.

Usually we don't copy-paste information from older releases, as the list "upgrade notes" would be very long if we include all previous versions in the list 🤷

lodesmets, 2024-05-31:
> > , so Aircoookie sees it maybe, and adds a warning to the documentation/ release note
> 
> @lodesmets the warning is already there, in the release notes for 0.13.3
> 
> [image]
> 
> The only news is that -since 0.14.0 - the old partition layout does not allow OTA update, as the new firmware is too big for the old OTA partition. So you need to update by USB, and make sure to have a backup of presets and config because these will be lost when updating.
> 
> Usually we don't copy-paste information from older releases, as the "upgrade notes" list would be very long if we include all previous versions 🤷 

Oh. I only looked in the 1.14 and 1.15 release notes. Sorry 😅

closed 2024-05-31 by lodesmets