wled/WLED #3989: Setting in the config to tell it what binary to grab

monsieurlatte, 2024-05-18:
**Is your feature request related to a problem? Please describe.**
Not really, if you could tell the device what binary to grab I could do the updates from within HA directly.  

**Describe the solution you'd like**
I have some athom devices that work perfectly fine but I have to manually grab the .zip for them and update manually.  

**Describe alternatives you've considered**
Nothing as it's not a huge problem just a "would be nice" feature :) 

**Additional context**
Appreciate the project itself and if you dont have time or interest in this I totally understand.

Thank you for your ideas for making WLED better!

blazoncek, 2024-05-18:
That's already available in 0.15

monsieurlatte, 2024-05-18:
hah, that's awesome and I hadn't gotten around to look at .15 release notes!

closed 2024-05-18 by monsieurlatte