wled/WLED #3990: Matrix Effect and Text Overlay interaction

PsalmEquipt, 2024-05-19:
### What happened?

The Matrix effect interacts with the text overlay, flashing brightly across the entire text. This makes an otherwise favorite effect unappealing to use and it is not adjustable with the controls. 

https://github.com/Aircoookie/WLED/assets/93995680/9bb72bb1-2a56-4a4a-bf55-4686ff97210d

### To Reproduce Bug

Select and display the Matrix effect using Segment 0. Overlay a text using the Scrolling Text effect on Segment 1.

### Expected Behavior

It would be more appealing to allow the Matrix Effect to remain in the background and not interact with the text or to allow that possibility with a control adjustment.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.4 Hoshi

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

I am enjoying WLED. Great job. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-05-19:
Use Drip or Rain which are similar effects.
Unforunately not all effects behave as expected when using overlays.

In this particular case Matrix effect needs to be modified. You can try 0.15 which has a bit different approach.

PsalmEquipt, 2024-05-19:
And when the day of Pentecost was fully come, they were all with one accord
in one place. Acts 2:1

Hi Blaž,

Thanks for understanding and for responding. I'm thrilled that overlays are
even a thing. I'm so glad I got WLED instead of a senior clock. Today I am
especially enjoying Frizzles.

Enjoy your evening,
Dave

* Dave Flores*
THE WAY Biblical Research, Teaching, and Fellowship

  Biblical Student in Akron, OH

  (234) 208-7012 <http://voice.google.com/calls?a=nc,%2B12342087012>
<http://voice.google.com/calls?a=nc,%2B12342087012>
<http://voice.google.com/calls?a=nc,%2B12342087012>
<http://voice.google.com/calls?a=nc,%2B12342087012>

  https://psalmequipt.nichesite.org <http://psalmequipt.nichesite.org>
<https://twitter.com/thewayintnl>
<https://www.flickr.com/photos/thewayinternational/albums>
<https://www.youtube.com/playlist?list=PLp3A63pN5wmzGpSzoQLtb_tmIhG5fyEU6>
<https://apps.apple.com/us/app/the-way-international/id1472818575?Is=1>
<https://play.google.com/store/apps/details?id=org.theway.twiapp&hl=en>
‌

dosipod, 2024-05-20:
@PsalmEquipt 
[image]
It is working fine , make sure you select  matrix  fx on the first segment and in the text fx select overlay and you shall have it .I think it should be okay in other versions but i mainly test now on latest 15 so try it out

blazoncek, 2024-05-20:
There is an issue with Matrix on anything below 0.15.0-b1

dosipod, 2024-05-22:
Looking closer to his video i think his unit is even under powered hence  i see octopus effect  looked bad

PsalmEquipt, 2024-10-21:
Thanks again for your efforts. I'm now enjoying a DIY WLED smart mirror. 
[image]

closed 2024-10-21 by PsalmEquipt