wled/WLED #3996: Effect mode_2Dcrazybees fails on larger matrices because int8_t for error2 goes OOB.

cschill2020, 2024-05-24:
### What happened?

The error2 variable can be greater than 255 for large matrices.  When the error2 rolls over the effect breaks and stays static (no more bees/flowers are generated). 

### To Reproduce Bug

Test run the effect on an 8x100 matrix.  You will see bee.posX and bee.posY can go OOB

### Expected Behavior

Easy fix is to make error2 int16_t for mode_2Dcrazybees:
https://github.com/cschill2020/WLED/commit/2c87f7e752527418eb4330a906a954e996568000

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.2

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

I was running on a dig-octa with 8x 100 pixel strips attached. 
You can find a fix here:
https://github.com/cschill2020/WLED/commit/2c87f7e752527418eb4330a906a954e996568000

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-05-24:
Thanks. Actually I am revisiting *all* effects to reduce the code and increase speed and one of the optimisations is the use of `unsigned` and `int` instead of 8 or 16 bit variants. I am not yet finished as some portions of the code do require variables to overflow (or wrap) and need to do further testing.

cschill2020, 2024-05-24:
Awesome!  For context I have been working on a port to run the WLED webapp
on Linux/osx.  I rewrote most of the http server and json/XML code using
libhttpserver and nlohman jason, and #defined out most of the Arduino
specific annotations (PROGMEM) etc...

I am running a 96x100pixel (30,000 ws2811 LED) matrix at BM using a
kulp-64D controller.  The webapp is able to run all 2d/1d/SR
effects(currently only using the sound simulator) and sends the led
instructions to the kulp using DDP.

I have been going through the effects 1-by-1 to test out on my current
8x100 setup.  E.g. add M12 mapping to BPM as it looks cool on a large
matrix cylinder.

I have a few ideas for adding new capabilities to existing effects.  I'll
keep some notes as I'm testing effects and will share them.  Happy to help
contribute if they sound broadly interesting/useful.

closed 2024-10-08 by github-actions[bot]