wled/WLED #4002: Esp8266-01

Johandigitaal, 2024-05-26:
### What happened?

Dear, I cannot get the esp8266-01 as well as the WemosD1 mini flashed with Wled.

### To Reproduce Bug

It will write but I can't get any further, I can't set anything, only the bug console appears

### Expected Behavior

Since I only need 1 datapin I want to use the esp8266-01 to control my ledstrip

### Install Method

Self-Compiled

### What version of WLED?

0.14.4

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2024-05-26:
@Johandigitaal what's the flash size of your device? 2MB, or 4MB?

Johandigitaal, 2024-05-26:
4Mb...

Johandigitaal, 2024-05-26:
The esp8266-01 have 1Mb

softhack007, 2024-05-26:
Which build environment did you use in platformIO?

softhack007, 2024-05-26:
> The esp8266-01 have 1Mb

1MB devices may not work (flash too small),  however your 4MB board should be ok if you use [`[env:nodemcuv2]`](https://github.com/Aircoookie/WLED/blob/2f6f0d2f3e3ca7c016d7f507a9a5433564484de0/platformio.ini#L329)

Johandigitaal, 2024-05-26:
for the Wemos D1 mini it gives the message : Failed to execute 'open' on 'SerialPort': Failed to open serial port.
Does see the board if you want to flash it ?

softhack007, 2024-05-26:
> for the Wemos D1 mini it gives the message : Failed to execute 'open' on 'SerialPort': Failed to open serial port. Does see the board if you want to flash it ?

That's a usually  driver problem (wrong/missing usb driver). 

Maybe your lucky and you'll just need to add `upload_resetmethod = nodemcu` to your custom build env.

Johandigitaal, 2024-05-26:
I removed and reinstalled the old driver ch340. Then I was able to flash the Wemos. The esp8266-01 fails as you mentioned earlier that the memory is too small.
Still thanks for helping me.mgv
Johan

dosipod, 2024-05-27:
@Johandigitaal A bit fiddly and i might have faced a bit of an issue flashing 0.14.4 not sure but i did not try hard enough  , still WLED Beta Release 0.15.0-b3 flashed okay on my esp01 with the gui flashing tool here and i can access the unit 
https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/WLED_%20ESP_Flasher
[image]

blazoncek, 2024-05-27:
Github is not the place to troubleshoot installation issues. WLED has been tested and works on every variant of ESP8266 unless there is a hardware issue with that particular ESP8266 (my own tests are done on 20+ ESP8266 devices, including ESP01, ESP01S, Wemos D1 mini and its clones).

Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.
Closing.

closed 2024-05-27 by blazoncek