wled/WLED #4003: web flasher seems to not be writing bootloader correctly

ancipital, 2024-05-26:
### What happened?

Using a 16MB ESP32-S3 feather, I can flash it with the https://install.wled.me/ - but then I seem to get a device that doesn't boot, so the AP never appears etc..

I tried manually flashing a bootloader after flashing with the web app, with:

`esptool.exe write_flash 0x0 .\esp32_bootloader.bin`

...using the esp32 bootloader that you provided in another issue thread. I also tried the "v4 bootloader", with identical results:

`esptool.exe write_flash 0x0 .\esp32_bootloader_v4.bin`

I am not very au fait with these devices, it's possible I had the wrong bootloader. Please speak up if that is the case!

I also tried flashing manually:

`esptool.exe write_flash 0x10000 WLED_0.15.0-b3_ESP32-S3_16MB_opi.bin`

I've tried adding either of the above bootloaders before or after flashing the main payload. I also erased before each "permutation", just in case, with:

`esptool.exe erase_flash`

No dice, whatever I tried. Observed behavior is that the serial port keeps coming and going, which is why I assume it's failing to boot with WLED on board. I'd love you to point out a neophyte mistake that I've made, if there is one. If there isn't, if there's a way I can help provide better info, please let me know.

It's worth noting that it works fine with ESPHome, flash and boots perfectly and joins the provisioned network ready to be adopted by Home Assistant. I was expecting that to be harder than setting up WLED :)

### To Reproduce Bug

Flash 16MB ESP32-S3 feather with https://install.wled.me/ tool. Try to find AP without success.

### Expected Behavior

ESP boots, and brings up an AP.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED_0.15.0-b3 (and others, all behave the same)

### Which microcontroller/board are you seeing the problem on?

ESP32-S3

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2024-05-27:
@ancipital This might not be true for your particular board but i used this https://wled-install.github.io/ to flash initially ( there is also a post from softhack on the discord with esptool )  after that i used OTA to test other builds 

[image]

blazoncek, 2024-05-27:
Please open an issue at WLED installer repository.

closed 2024-05-27 by blazoncek

softhack007, 2024-05-27:
> using the esp32 bootloader that you provided in another issue thread. I also tried the "v4 bootloader",

-S3 (and also -S2, -C3) need their own bootloader, they won't  start with a bootloader for esp32.

dosipod, 2024-05-27:
@softhack007 Hey , I recall the post(s) you did on discord for s3 ,s2 and C3 . Even if the installer works for some boards it might not works for others so as we see a lot of guys having issues like this we tried to get them to contact you on github , might be not worth opening an issue for flashing ( Other repos enabled discussion on github  so might be that could help as closed issues is sometime missed ) .Otherwise I was thinking to retest again once we could get more boards as we have only lolin s3,s3 and c3 and might be repeat your post  ( with your help)  to again setup a package for each board ( Will cost us a lot if you have any please spare us  )

ancipital, 2024-05-27:
> @ancipital This might not be true for your particular board but i used this https://wled-install.github.io/ to flash initially ( there is also a post from softhack on the discord with esptool ) [...]

No it doesn't. This is exactly why I haven't opened an issue on the  wled installer repo (which looks pretty quiet). None of the installers, the "official" WLED flasher, https://wled-install.github.io/ or using ESPtool produce a working install, no matter which of the apparently relevant firmwares I use, which makes me suspect that it's a wider WLED issue, and kneejerk closing this is premature.

Worth noting, as above, that it works every single time with the ESPhome installer, so the hardware isn't faulty either.

If there's a way to get @Aircoookie better information, happy to do so. Admittedly, I was hoping for a working firmware and documentation, and not having to go lurk on yet another Discord- it's not really a viable replacement for project docs :)

blazoncek, 2024-05-27:
Using esptool produces working WLED installation. So not a WLED issue per se.

softhack007, 2024-05-27:
... and yes, you should enter our [WLED discord](https://discord.gg/QAh7wJHrRM). 

WLED is developed by "a few good men", however we are all doing it in our free time and there are simply not enough people available to program new features, fix bugs _and_ maintain an exhaustive and up-to-date set of documents. 

On discord we have a very active user community, and very likely you'll either find the answer to your problem by searching previous messages, or you can ask in one of the channels (however no double posting!), and an experienced user might have the time to help you.

GitStorageOne, 2025-02-13:
Unfortunately, WLED cannot boot up on [ESP-S3 N16R8 (ESP32-S3-DevKitC-1)](https://www.amazon.com/dp/B0CKXJLP4B). 
I can share what I have:  

```
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
---
Manufacturer: 85
Device: 2018
Detected flash size: 16MB
Flash type set in eFuse: quad (4 data lines)
```

Assume, issue somewhere inside esp32_bootloader_v4.  
Is there any way to get it work?

[image]

GitStorageOne, 2025-02-13:
I tried https://wled-install.github.io/

[image]

Works just fine.  
So, why WLED manual and https://install.wled.me/ does not provide same experiences?

GitStorageOne, 2025-02-13:
I extracted bootloader from https://wled-install.github.io firmware (0x0 0x9999) and then uploaded original `WLED_0.16.0-alpha_ESP32-S3_16MB_opi.bin` (0x10000 ...).  
It works. So, `esp32_bootloader_v4.bin` is not proper bootloader for ESP32-S3.