wled/WLED #4005: Full brightness on LED count change 2D Matrix

Blowntobytes, 2024-05-28:
### What happened?

When changing LED count for a 2D matrix, the panel goes to full brightness upon saving.

### To Reproduce Bug

Simply change LED count. I went from 256 to 255 and panel went to full bright on saving.

### Expected Behavior

Was hoping the panel would stay the same brightness as the slider indicated and from previous setting.

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0-b3

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

freakintoddles2, 2024-06-02:
@blazoncek this is a really annoying issue with 15.x builds, would be nice to have it addressed. It happens for me when changing seemingly anything in the led settings.

blazoncek, 2024-06-03:
How many times do you change LED count?
Sorry, but backburner.

DedeHai, 2024-10-02:
I looked into this bug as it is really annyoing. it changes the brightness every time the settings are saved, not only when ledcount changes, any change to LED settings triggers a full brightness.
here is what I found so far:
the global `bri` variable is not changed, it stays at the set level.
I hunted down where the bug was introduced but was unable to find which commits were pulled in with the merge-commit in question:
in commit acf6736afdc579e0bb498b9111e2b8ff7cbaab63 it still works, next commit 7c8df9796868b045da08f92830520f4bd3cda656  introduces this bug.

blazoncek, 2024-10-02:
Have you checked different NPB versions?

DedeHai, 2024-10-02:
not yet. I first need to figure out what happened in 7c8df9796868b045da08f92830520f4bd3cda656 as the commit itself does not introduce any new code, but since its a merge commit it probably pulled a lot of stuff in right? 
how can I compare full code?

blazoncek, 2024-10-02:
That was just a merge of main branch to make it compatible for the opposite direction merge later on.
That did not bring anything bus related back into 0_15.

DedeHai, 2024-10-02:
> That did not bring anything bus related back into 0_15.
if you have the time, try this:
- check out the commit acf6736afdc579e0bb498b9111e2b8ff7cbaab63 (which is the last one that works). upload and go to LED settings: change any settings (i usually just change the value of the ABL) and save. Nothing bad happens
- Now check out the next commit in the history 7c8df9796868b045da08f92830520f4bd3cda656 which is the merge commit. Repeat the steps of changing something, save. LEDs (adressable ones) will go full brightness. 
I still need to check if the ABL is also disregarded in this bug. If it is, this definitely must be fixed before 0.15 release.

DedeHai, 2024-10-02:
that would be an easy fix. I can test if that does the trick and it may very well work, as the brightness is fixed once I move the slider in the UI.
so in that merge commit: was the NPB updated to a newer version or how is it that all of a sudden this bug appears?

blazoncek, 2024-10-02:
Bug found and mitigated.

DedeHai, 2024-10-02:
that was quick, thank you! 
Did you figure out what change caused it?

blazoncek, 2024-10-02:
yes, re-initialised NPB

closed 2024-10-02 by blazoncek