wled/WLED #4007: Websocket response when changing the parameters Target brightness and Target time when setting the timer

VolchkovVlad, 2024-05-29:
### What happened?

When sending parameters such as nl.tbri and so nl.dr WLED responds with {"success":true}

### To Reproduce Bug

Connect to WLED via WebSocket, send {"nl":{"tbri":255}} or {"nl":{"dur":3}}

### Expected Behavior

[image]

### Install Method

Binary from WLED.me

### What version of WLED?

"0.15.0-b2" build 2404100

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

Output from POSTMAN

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-05-30:
You need to add "verbose" command, `"v":true`.

closed 2024-05-30 by blazoncek

VolchkovVlad, 2024-05-30:
"v":true will simply return the current state. I have no problem getting the current state. I'm just pointing out the inconsistency of the documentation

blazoncek, 2024-05-30:
You are missing the crucial detail.
When you send JSON API via websockets and do not include `"v":true` you will only get `{"success":true}` or `{"error":X}` as a response (in that session) but a websocket broadcast (to all clients) that will follow will include state and info.

VolchkovVlad, 2024-05-30:
No, I am sending data without "v:true". In response, I receive /json/si. Again, the documentation says that you need to use v:true only to provide additional information. Nowhere does it say that this is a mandatory key

blazoncek, 2024-05-31:
You are welcome to update documentation if you feel it is incorrect.

VolchkovVlad, 2024-05-31:
If there is such a problem with people, we can discuss it. But in my opinion, developers should write documentation

blazoncek, 2024-05-31:
If you pay me, perhaps I will. Or someone else for that matter.
Until then WLED is a collaboration project where people contribute out of their free will without any compensation.
That is also why you get to use (and abuse) WLED for free.

VolchkovVlad, 2024-05-31:
What does the payment have to do with it? I just pointed out an inaccuracy in the documentation. (go to the data exchange protocol) and that's it. I don't demand to fix it urgently.  I'm just sharing what I saw, maybe it will help the development of the project.

I have bypassed this problem, it does not bother me.