wled/WLED #4015: ledcDetachPin code not working with  ESP32 3.0.0 core.

mrdude2478, 2024-06-05:
### What happened?

Unable to com compile with  ESP32 3.0.0+ core. 

### To Reproduce Bug

Update esp32 boards from 2.7 to new 3.0+ cores

### Expected Behavior

Wanted to compile, but fails on new cores dues to missing ledcDetachPin code in new libraries.

### Install Method

Binary from WLED.me

### What version of WLED?

Latest release from git release page.

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
Not needed.
```

### Anything else?

I spent about 4 hours trying to figure out why this wouldn't compile, then I found that someone had updated the esp32 board to the newer 3.0 core, after downgrading back to 2.7 core I was able to compile again.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-06-06:
Nobody said WLED is developed for core 3.0.0 (whatever that is).

Please specify what core 3.0.0. means to you.

softhack007, 2024-06-06:
> Unable to compile with ESP32 3.0.0+ core.

I you think about using arduino-esp32 v3.0.x - it's not possible.
* not supported in platform-espressif32 (see this thread: https://github.com/platformio/platform-espressif32/issues/1225)
* long list of breaking changes for arduino-esp32 (ledc is only one of them) https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html
* even longer list of breaking changes in ESP-IDF v5: https://docs.espressif.com/projects/esp-idf/en/v5.1.4/esp32/migration-guides/release-5.x/5.0/index.html and https://docs.espressif.com/projects/esp-idf/en/v5.1.4/esp32/migration-guides/release-5.x/5.1/index.html
* libraries we need must be ported to arduino core v3.0.x, too

As espressif have changed a lot of functions and basicially nothing works s before, there would be a lot of work just to "upgrade" to the latest arduino-esp32.

softhack007, 2024-06-06:
> Update esp32 boards from 2.7 to new 3.0+ cores

PS are you trying to build WLED in Arduino-IDE? Please use VSCode with platformIO. https://kno.wled.ge/advanced/compiling-wled/

Building WLED in Arduino-IDE is not supported any more.

mrdude2478, 2024-06-07:
Thanks, I've installed visual studio 2022 and Platformio so I'll compiled with that instead, I can compile fine on Arduino though if I disable the a Alexa code and use the 2.7 boards core (I don't use Alexa anyway).

closed 2024-06-07 by mrdude2478