wled/WLED #4016: pxmagic.htm error and cannot create gif with animation feature

bangnguyendev, 2024-06-07:
### What happened?

When using GIF to create a preset, an error and cannot be created. 
While using REPO https://github.com/ajotanc/pixelmagictool, it works well!

[image]

### To Reproduce Bug

When using GIF to create a preset, an error and cannot be created. 
While using REPO https://github.com/ajotanc/pixelmagictool, it works well!
[image]

### Expected Behavior

Use the feature to create a good active image!

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.4

### Which microcontroller/board are you seeing the problem on?

ESP8266, ESP32

### Relevant log/trace output

```shell
None log
```

### Anything else?

 None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2024-06-08:
@bangnguyendev You might want to report such issues on the owner gh . Still i was able to use a gif on 32x32  ( i got some initial errors as i am testing with 0.15.0-b3 but it kinda works on esp32 but might have issues .This whole solution is not optimal for gifs 

The good news is that you do not need to mess around with that html page anymore , check the new gif branch  https://github.com/Aircoookie/WLED/tree/gif 
[image]

blazoncek, 2024-06-08:
Tagging @ajotanc as he is the author of PixelMagic tool.

bangnguyendev, 2024-06-08:
I use @ajotanc Repo and it works good.

However, when used in WLED's pxmagic folder it doesn't work well

ajotanc, 2024-06-14:
> I use @ajotanc Repo and it works good.
> 
> However, when used in WLED's pxmagic folder it doesn't work well

I don't remember very well if after it was set as standard in WLED and I updated the project's GH, but if so, I'll look at the differences to send the corrections.

closed 2024-10-20 by github-actions[bot]