wled/WLED #4019: Segment Names not preserved across Reboot

askpatrickw, 2024-06-09:
### What happened?

I have a working configuration with the default segment names, including two presets.

I renamed the segments to have useful names (like room names).

I rebooted and the names reverted back to the default ones.

### To Reproduce Bug

I have a working configuration with the default segment names, including two presets.

I renamed the segments to have useful names (like room names).

I rebooted and the names reverted back to the default ones.

### Expected Behavior

I clicked save as I renamed each segment.
They should be saved and preserved across reboots.

If there is some other step that needs to be done, then the UI should tell me that.

### Install Method

Binary from WLED.me

### What version of WLED?

latest, has been going on for awhile

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

askpatrickw, 2024-06-09:
Is anyone using Segments in the manner this FAQ outlines?
https://kno.wled.ge/basics/faq/#my-segments-are-gone-after-a-reboot

softhack007, 2024-06-09:
Technicially speaking, segments are volatile. 

The object called "segment" gets deleted and replaced constantly. If you want to preserve segments after reboot, you need to save them in a preset. 

This works well for me. I'm frequently combining several effects on a LED matrix, and segments allow to to "partition" the matrix as necessary. When I'm happy with the result, I'll save a new preset. So yes, I'm using segments in the way described in the KB.

softhack007, 2024-06-09:
> I rebooted and the names reverted back to the default ones.

You probably use "make a segment for each output"?

softhack007, 2024-06-09:
> will be able to save multiple segment configurations

Not sure what you mean - this is exactly what happens when you hit "create preset". So you basicially want the same button replicated on the segments page?

closed 2024-06-09 by blazoncek

askpatrickw, 2024-06-09:
I'm trying to politely point out that while some may understands how this works, it is not at all obvious, intuitive or discoverable from the UI how this works and what the intended behaior is.

For a normal user (even one with a lot of software industry experience) when  you make a change, click save and then it is not saved. Because you had to go do some other change. That is hard to figure out, especially given the tabbed layout of the mobile UI.

Maybe WLED is just going to have a steeper learning curve. I mean you do need to do soem complicated electrical stuff anyway, so mayvbe that is fine.  But, I hope this issue existing so people can find it will help others as well discover how it should work, even if it isn't fixed. (It would have helped me.)

blazoncek, 2024-06-10:
By all means, please contribute to the documentation if you feel it is lacking.