wled/WLED #4021: Time wrong after some day

emmexx, 2024-06-13:
### What happened?

Sometimes, usually I notice the problem in the morning, the time of wled is wrong, around one day behind.
It happened 3 times since I installed 0.14.1 ESP8266 at the beginning of March 2024.
After rebooting the time was set correctly.

### To Reproduce Bug

I use the Analog clock overlay and NTP from 0.wled.pool.ntp.org
WLED is wifi connected to a router and the router is connected to internet using my phone hotspot. So there is no internet connection when I'm not home. But I'm home for many hours every day. I didn't leave the house for a month expecting to find the clock with the correct time.

### Expected Behavior

Time should be in sync with the value downloaded from the ntp server.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED v0.14.1 (build 2401141)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

emmexx, 2024-06-13:
@blazoncek My doubt was that after a number of failed attempts to connect to the NTP server, WLED stops doing it and so it has to rely on the ESP8266 RTC clock.
If that is the case I can set a preset to reboot it every night.

blazoncek, 2024-06-13:
WLED requests NTP time approximately twice per day, retrying unsuccessful request every few seconds.
If the NTP library fails to process retries correctly is is an issue of the library, not WLED.

closed 2024-10-19 by github-actions[bot]