wled/WLED #4027: Static IP address set but not reachable

jezzalencko, 2024-06-17:
### What happened?

Whenever I try to set a static IP on a VLAN (including static gateway), I restart and it seems to lock in that IP address successfully (visible on my network), however the IP is unreachable and I can't ping it.
And since there is no DHCP fall back, I have to disconnected everything and reflash just to be able to get access to the board again.

### To Reproduce Bug

I've tried the same on both an EST-PoE-32 and QuinLED-dig-Uno board with the same results.

### Expected Behavior

Should be able to change static IP address and access the new static address with that new IP from the web UI.

### Install Method

Binary from WLED.me

### What version of WLED?

tried both 14 and 15 (beta) for the EST.

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

As a minimum (if I can't get this static issue resolved), could I potentially set it to reference the DHCP of my IoT VLAN which is 192.168.20.x?
Currently it defaults to the DHCP of my main LAN subnet 192.168.0.x which initially connected.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-06-17:
You can always use a button to switch on AP mode.
Press and hold it for 6s or more.

blazoncek, 2024-06-23:
Possible duplicate of https://github.com/Aircoookie/WLED/issues/3817
Closing.

closed 2024-06-23 by blazoncek

jezzalencko, 2024-06-24:
> You can always use a button to switch on AP mode. Press and hold it for 6s or more.

Could you please elaborate on this? Do you mean just hold the reset button down on the EST-PoE-32 board?

blazoncek, 2024-06-24:
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.
See [KB](https://kno.wled.ge) as well.