wled/WLED #4031: Improv over serial shows duplicate network names

frenck, 2024-06-19:
### What happened?

I have multiple access point broadcasting the same SSID, however, the list provided over Improv seems to be based on BSSID.

[image]

It should be de-duplicated.

### To Reproduce Bug

Install WLED via install.wled.me, during network setup ensure you have at least two different BSSIDs broadcasting the same SSID.

### Expected Behavior

To have a clear and concise list to pick from.

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0-b3

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-06-19:
To minimize code needed (for one time job) SSIDs are displayed as they are discovered (see `improv.cpp` `handleImprovWiFiScan()`).
Deduplicating would require additional parsing and the code size would increase without any real benefit to WLED or end-user. Code size started to matter as we are hitting close to 100% flash usage (application partition).

I will admit that UX would perhaps be better with just a single/unique SSID but otherwise there are no benefits or drawbacks.

And BTW I would not call that a bug, perhaps enhancement.

frenck, 2024-06-19:
> without any real benefit to WLED or end-user

🤷 That I disagree with; this is a UX issue. I've looked at other applications using Improv, and concluded they do nicely deduplicate this 🤷  (I original thought it was an ESP Webtools issue).

> And BTW I would not call that a bug, perhaps enhancement.

I'm not fully in agreement, but I 🤷 no strong feelings about the classification either.

../Frenck

blazoncek, 2024-06-19:
I understand and I am just stating the reasons why nobody bothered.
I myself have similar situation (regarding multiple SSIDs and BSSIDs, but not Improv as I do not use it) so am aware of the underlying issue. In fact that was a boon when I implemented multiple WiFis a few months ago (in #3709 ).

Another option would be to sort and deduplicate SSIDd in web browser when displaying those.