wled/WLED #4032: Athom 15w led bulb fade on startup

koen-serneels, 2024-06-21:
I have a Athom 15w led bulb and flashed the latest stable 0.14.4. When the bulb starts ootb it has a warm white color with low brightness. I created a preset for the color I actually want and configured it to startup using that preset. However, at cold start the bulb first seems to show to the default color (it does that instantly without fading or anything) and like a couple of hundred msecs later it seems to change to the preset, this change is done using fading. This gives a bit of a clunky effect where the led instantly turns on to a non desired light setting and then starts to gently fade to the desired color. Is it possible to let the led start completely off and directly start with fading to the configured preset?

blazoncek, 2024-06-22:
Please set the transition time to 0 for boot preset.

blazoncek, 2024-06-26:
Please close the issue if the workaround works, otherwise please provide an update.

Andro-Marian, 2024-07-01:
Not working for my. I set the "Turn LEDs on after power up/reset" to off, "Default brightness" to 0, Preset to 1.

Still wakes up as Orange color then starting to change to red.

`{"on":true,"bri":128,"transition":0,"mainseg":0,"seg":[{"id":0,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"col":[[255,0,0,0],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0}]}`

Jusius, 2024-07-02:
actually this is happened to me when i install from the bin file
because the bin configuration set the start color with orange on the first boot 

even when change the transition time set to 0 it still start with orange then change to preset 

you have to compile your own code to make the start color different

blazoncek, 2024-07-02:
Duplicate of #3196

closed 2024-10-27 by blazoncek

Andro-Marian, 2024-10-31:
I flashed the 0.15.0-b6 and when turning off/on the bulbs (7w or 15w) has a weird flicker when using the power putton from the web interface. Not sure why.