wled/WLED #4034: Implement phase shifted PWM

blazoncek, 2024-06-27:
Implement phase shifted PWM outputs to lower the demand on PSU and/or allow 2-channel [polarity changed LED](https://www.amazon.de/BESCOST-Lichterkette-Strombetrieben-Hochzeit-Weihnachten/dp/B0BBQS17LW/) strips to function properly.

The discussion about (issues with) PWM started in #3962 as a result of crashes within PWM handling code.

Requested participants: @softhack007 @willmmiles @jw2013 @tablatronix and (optional) @Jason2866 
Please follow up here from https://github.com/Aircoookie/WLED/pull/3962#issuecomment-2194289342

References:
https://www.hackster.io/daniel232/esp32-generate-two-180-phase-shifted-pwms-c1010f
"On ESP8266, there is a hidden, but officially supported solution for PS-PWM" by @jw2013

willmmiles, 2024-09-28:
Phase shifting on ESP32 was prototyped in #4115 and merged in #4129.

blazoncek, 2024-12-05:
I am content with current implementation.

closed 2024-12-05 by blazoncek

lk-ek, 2025-07-12:
Just here to say thank you for your efforts! This works like a charm and made a friend of mine very happy as their opposite-polarity-fairy lights are now dimmable and controllable via their mobile :)