wled/WLED #4040: Incorrect ADC Readings Due to 13Bit capture width as defaults on ESP32-S2

Orel-A, 2024-07-04:
### What happened?

handleAnalog() function incorrectly assumes the analogRead() arduino API on ESP32-S2 will give a value between 0-4095 but in fact it's 0-8191 (13bit)

### To Reproduce Bug

-

### Expected Behavior

-

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0-b4

### Which microcontroller/board are you seeing the problem on?

ESP32-S2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-07-05:
> Can be fixed with a call of analogReadResolution(12); on init code or change the divide by 16 in handleAnalog() function.

You are welcome to provide a PR.

blazoncek, 2024-07-07:
You can try latest 0_15.

Orel-A, 2024-07-07:
The fix is working, thanks.

closed 2024-07-07 by Orel-A