wled/WLED #4052: TKIP is used as group cipher and offered as pairwise cipher.

Frostie314159, 2024-07-11:
### What happened?

On the event where I am right now, there are a lot of ESPs with WLED. Looking at the Wi-Fi beacons with monitor mode reveals, that all of them use TKIP as the group cipher suite and offer it optionally as a pairwise cipher suite. TKIP has been deprecated for over a decade by now.

### To Reproduce Bug

Use WLED and observe the beacons in monitor mode. I have a pcap file with the beacon frame.

### Expected Behavior

Disable TKIP entirely and use just CCMP. Possibly even enable management frame protection optionally.
[wled_cap.pcapng.gz](https://github.com/user-attachments/files/16182802/wled_cap.pcapng.gz)

### Install Method

Binary from WLED.me

### What version of WLED?

all

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Anything else?

I'm unsure about the specific revisions of the boards and software used, since I'm just looking at beacons.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-07-11:
Googling around revealed no information on how to change cipher from TKIP to CCMP.
It would be helpful to provide some information on how to do that.

Frostie314159, 2024-07-11:
What library are you using? Esp wifi should be capable of doing this.

blazoncek, 2024-07-12:
> What library are you using? Esp wifi should be capable of doing this.

Can you provide a sample? Using Arduino 2.0.9 framework using standard/embedded WiFi library.

softhack007, 2024-07-12:
Found one hint
* https://github.com/espressif/arduino-esp32/issues/4805

blazoncek, 2024-07-13:
Thanks @softhack007 I've seen it mentioned in framework sources but no example on how to set it up.
Following through issues it looks like this was changed/implemented in Arduino core 2.0.0. And only for ESP32.

@Frostie314159 have you tried 0.15 with ESP32 or just 0.14 (or below) and ESP8266?

I am inclined to close this issue as it is not related to WLED but rather Arduino implementation of SoftAP.

Frostie314159, 2024-07-13:
Yeah, this can be closed, I'm going to proceed with investigating this on the Arduino side.

closed 2024-07-13 by Frostie314159

Aircoookie, 2024-07-15:
OK, this sucks. TKIP is broken and I can't wrap my head around why there isn't an option to disable it if AES is supported on ESP8266...
See https://github.com/esp8266/Arduino/issues/7825 and https://github.com/espressif/ESP8266_NONOS_SDK/issues/351

For clarification, this insecure cipher affects ESP8266 in AP mode. ESP32 uses CCMP only today, as it should be.

Frostie314159, 2024-07-15:
I've already opened an issue on the esp8266 Arduino core repo, but haven't received a response yet. The code is also extremely hard to read, so I wasn't able to trace the source myself.