wled/WLED #4053: Wled Usermods BME280 HomeAssistantDiscovery

Yoanf26, 2024-07-12:
### What happened?

Good morning,

I have a problem with reporting information from the BME280 sensor or I only have the temperature and pressure reported in Home Assistant. Except I have the temperature, pressure, humidity, dew point and heat index which are clearly indicated on the WLED page.

[image]
[image]

### To Reproduce Bug

Compile modified firmware with the BME280 v2 usermod, connect a BME280 sensor, activate MQTT, publish always and Home Assistant Discovery

### Expected Behavior

Have all the sensors reported in Home Assistant

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.14.4 (Build 2405180)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-07-12:
Please tag usermod author or last contributor as WLED team does not maintain usermods.
As far as I can see it should be @albarlow .

Yoanf26, 2024-07-12:
Sorry. I did not know. At the same time, I am trying to find out why only the temperature and pressure rise and not the humidity.

I was able to retrieve the values ​​via Home Assistant, but it would be better if this was done automatically.

softhack007, 2024-07-12:
> At the same time, I am trying to find out why only the temperature and pressure rise and not the humidity.

This could be normal, because one of the two models (not sure if BME280 or BMP280) does not measure humidity.

See https://www.ourpcb.com/bmp280-vs-bme280.html

Yoanf26, 2024-07-12:
No it's not normal. The usermod is for the BMP280 AND the BME280.

Look at the screenshots, it recognizes the BME280, and on the MQTT-Explorer screenshots, we can see that they are there too. Except that HomeAssistantDiscovery doesn't work with humidity, dew point and heat index. However, it is present in the usermod script, except that it gets stuck.

softhack007, 2024-07-12:
Maybe the usermod author can help.

Isn't it great that a firmware made to control LEDs can be abused as a weather station 😉

Yoanf26, 2024-07-13:
It's not for a weather station. I did one and handled it differently. But I am making a Japanese lamp for my partner's daughter  in 3D printing and to avoid having two ESP8266 modules (one for the Japanese lamp and one to have the temperature and humidity information for heating management via Home Assistant), I wanted to unify everything. Except that only the temperature and pressure are published in HomeAssistantDiscovery and in parallel, everything is published in MQTT outside HomeAssistantDiscovery. I know that I can add the missing information manually in the configuration.yaml of Home Assistant (That's what I did while waiting for it to work in HomeAssistantDiscovery)

softhack007, 2024-07-13:
It looks like the BME280 usermod wasn't updated for about 3-4 years, so maybe the original author left if behind. 

WLED is a community project - as core maintainer we usually don't have the time to care for all usermods, so if you have a need, maybe take it as a chance to add something to WLED, and start a PR to fix the BME280 usermod.

Or alternatively, try other another temp&humidity sensor like the [sht31](https://github.com/Aircoookie/WLED/tree/0_15/usermods/sht#mqtt-discovery-for-home-assistant), [dallas](https://github.com/Aircoookie/WLED/tree/0_15/usermods/Temperature) or [BME680](https://github.com/Aircoookie/WLED/tree/0_15/usermods/BME68X_v2).

closed 2024-11-20 by github-actions[bot]