wled/WLED #4055: Brightness change per json api results in unresponsiveness

tj0vtj0v, 2024-07-14:
### What happened?

I want to change the brightness per json api 
requests.post(http://192.168.178.57/json/, {"bri": 1})
which isn't working and even leading to no reaction when using the slider in the Web-UI.

### To Reproduce Bug

requests.post('http://<IP-Adress>/json/', '{"bri": 1}')
in Python

### Expected Behavior

change brightness to 1 (gamma-correction is off)

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.2 - 2403170

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-07-14:
When reporting errors, use latest available version. Please.

dosipod, 2024-07-16:
@tj0vtj0v 

This is a Python script  i used to test with and it worked on older builds , i used the same now on the latest 0.15.0-b4 and it worked as expected . Please use that as you could be doing something wrong in your test (  Syntax in Python  to use json is a bit different) 
[py_script_01.txt](https://github.com/user-attachments/files/16243528/py_script_01.txt)

closed 2024-08-28 by tj0vtj0v