wled/WLED #4057: Clock preset on Matrix display rendered out of bounds

Plawasan, 2024-07-15:
### What happened?

I'm using a WS2812b 32x8 matrix display with WLED 0.15.0-b4. I'm using the Scrolling Text present with "#HHMM" as the segment title to display a clock. This has always been working perfectly fine however in the latest beta, the text gets sometimes misaligned:

[image]

When I then switch to a different preset and back to my "clock" preset, it's rendered correctly:

[image]

LED Config:

[image]

2D config:

[image]

### To Reproduce Bug

Create a Scrolling text preset, update segment name to #HHMM 
Wait?

### Expected Behavior

Display entire text centered

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0-b4

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
Preset:

   "14":{
      "on":true,
      "bri":65,
      "transition":7,
      "mainseg":0,
      "seg":[
         {
            "id":0,
            "start":0,
            "stop":32,
            "startY":0,
            "stopY":8,
            "grp":1,
            "spc":0,
            "of":0,
            "on":true,
            "frz":false,
            "bri":255,
            "cct":127,
            "set":0,
            "n":"#HHMM",
            "col":[
               [
                  255,
                  0,
                  0
               ],
               [
                  0,
                  0,
                  0
               ],
               [
                  0,
                  0,
                  255
               ]
            ],
            "fx":122,
            "sx":128,
            "ix":155,
            "pal":0,
            "c1":0,
            "c2":128,
            "c3":16,
            "sel":true,
            "rev":false,
            "mi":false,
            "rY":false,
            "mY":false,
            "tp":false,
            "o1":true,
            "o2":false,
            "o3":false,
            "si":0,
            "m12":0
         },
         {
            "stop":0
         },
         {
            "stop":0
         },
         {
            "stop":0
         },
         {
            "stop":0
         },
         {
            "stop":0
         },
         {
            "stop":0
         },
         {
            "stop":0
         },
         {
            "stop":0
         },
         {
            "stop":0
         },
         {
            "stop":0
         },
         {
            "stop":0
         },
         {
            "stop":0
         },
         {
            "stop":0
         },
         {
            "stop":0
         },
         {
            "stop":0
         },
         {
            "stop":0
         },
         {
            "stop":0
         },
         {
            "stop":0
         },
         {
            "stop":0
         },
         {
            "stop":0
         },
         {
            "stop":0
         },
         {
            "stop":0
         },
         {
            "stop":0
         },
         {
            "stop":0
         },
         {
            "stop":0
         },
         {
            "stop":0
         },
         {
            "stop":0
         },
         {
            "stop":0
         },
         {
            "stop":0
         },
         {
            "stop":0
         },
         {
            "stop":0
         }
      ],
      "n":"Clock"
   },
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2024-07-16:
@Plawasan 
I recall something on older builds when loading presets with scrolling text . But testing now on latest from here https://github.com/dosipod/PUB-WLD-GH-COMILE/tree/main/.github/workflows
My 8x32 setup works as expected and i see the time with scrolling text centered as you see in the example
with 5 leds space on one side and 4 on the other side ( that would change if you change the sliders ) 
[image]

Please check the Y offset slider ( and other sliders to be centered  ) before saving presets .If you did that then 
clarify more as saying "sometimes" means at random and not sure how to test that as i did save a preset with 
text and it worked as expected .The same was also true on 32x16 and also on 32x8 segment created on the 32x16 . 
If all fails then might be reset your unit ( after backup ) and verify further

Plawasan, 2024-07-17:
I haven't seen the original issue in 24h but there's still something inconsistent about the alignment of the text. This is how the clock came up this morning after being switched off for the night, notice it's aligned too far left:

[image]

And then this is what it looked like after I've simply switched to a different preset and back to the default one (the one I posted above) - now it's correctly centered:

[image]

Btw my Y Offset it as 155 but that affects position along the Y axis (up/down), not left to right.

dosipod, 2024-07-17:
It is true the Y offset will change the up/down , i just wanted all the sliders to be default . The only difference i can see is you used a bin  from WLED.me   . So not really sure how to replicate that ( I actually tried to force is to have an offset but could only do that by adding an offset in the 2D page which clearly not the case with you ).

If you reset the unit and used the bin from the link i listed then we could still try to keep your preset for few days but you have wrongly pasted that as it is truncated at the end   , save it to text file  and add it here and we could try .

Plawasan, 2024-07-17:
Humor me, where do I find the binary you want me to test (I only see yaml)? The .bin I'm using came from the release page for the latest beta on the WLED github page.

I've pasted only the relevant preset however my full config and all presets are attached.
[wled_cfg_WLED-Clock.json](https://github.com/user-attachments/files/16262209/wled_cfg_WLED-Clock.json)
[wled_presets_WLED-Clock.json](https://github.com/user-attachments/files/16262210/wled_presets_WLED-Clock.json)

dosipod, 2024-07-17:
The run i used the bins from are here https://github.com/Aircoookie/WLED/actions/runs/9724727410/job/26841257355

Plawasan, 2024-09-11:
Issue still present on 0.15.0-b5, switching to a different preset and then back to text fixes it:

@dosipod I might be a bit slow but I still don't see a binary I could use on the link you posted, only the progress of the build..

Plawasan, 2024-09-23:
> @Plawasan The trick is to first click on "summary", then scroll down the right panel until you see "Artifacts".

Thanks, got the binary, updated my WLED instance (or rather downgraded back to 0.14.4) and I have not seen the issue come up since then.

Is this binary a fork with a specific fix that has an effect on this or is that one of the official release builds? I'll switch back now to the latest 0.15 beta to see if the issue comes back.

softhack007, 2024-09-28:
> Is this binary a fork with a specific fix that has an effect on this or is that one of the official release builds? I'll switch back now to the latest 0.15 beta to see if the issue comes back.

@Plawasan it's a 0.15 development build (latest work in progress). Do you still see the problem in the 0.15 beta ?

Plawasan, 2024-09-28:
> @Plawasan it's a 0.15 development build (latest work in progress). Do you still see the problem in the 0.15 beta ?
 I'm still on the custom build and still working perfectly fine, I wanted to run it for a while given the intermittent nature of the problem.. I'll go back to the beta but if 0.15.0-b4 is still the latest then that's where I saw the issue.

blazoncek, 2024-09-28:
No matter what I do, I cannot replicate.

Plawasan, 2024-09-28:
Updated to v0.15.0-b5, will report back if it shows up again

closed 2025-02-04 by github-actions[bot]