wled/WLED #4059: ESP32 dies after a few hours of running

hoanglam9925, 2024-07-16:
### What happened?

I installed wled into the esp32, I didn't install the led, just plugged it in and hung the chip to run the code to try, without attaching any peripherals, but it died after a few hours of hanging the code, it couldn't recognize the COM port and also There is no more website to configure.

### To Reproduce Bug

Install and wait few hours

### Expected Behavior

How to prevent chip died

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.4

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2024-07-17:
@hoanglam9925 You are having a HW issues as wled will not kill a chip in few hours or few years . Please use discord for such issues

OxyDroiD, 2024-08-08:
Hi. I tried what you tell us. My wled working fine. My esp is ESP32-Wroom-32d. Maybe you connected too many leds in esp's 5v pin or you use 12v led and drive led's data using esp?

softhack007, 2024-09-28:
Closed as no developer was able to reproduce this problem.
Could be an electrical issue (wiring).

closed 2024-09-28 by softhack007