wled/WLED #4062: Button press would erase configuration and presets

flux242, 2024-07-19:
I have soldered a touch button to the ESP01. As I configured the pin and Push type. The first time I touched the button it erased all my stored settings. After some research is was clear to me that I should have used PushInverted type of button because wled expects that a button would pull pin to the ground. But after restoring configuration I tried to change the button type to Inverted and upon saving it would still erase my settings again because I couldn't switch into the settings screen within 12 seconds after wled booting!!! So, it is not possible to restore configuration from a file and change button setting in the GUI! And this is anything but well defined UI

So, to prevent this from happening one could reset the button timer upon configuration changes
Alternatively one could expose button timers in the GUI, so that one could disable/enable 6/12 sec timers or even redefine its actions

blazoncek, 2024-07-19:
Assign your GPIO to button 1.

flux242, 2024-07-19:
really? How should it help in the situation described?

blazoncek, 2024-07-19:
Use custom compilation.

flux242, 2024-07-19:
really? Desolder the button instead of resetting the timer in software? Sounds like a plan

blazoncek, 2024-07-19:
Do some prototyping 1st, then solder.

Or use my other proposal.

softhack007, 2024-07-19:
> really? Desolder the button instead of resetting the timer in software? Sounds like a plan

In fact, it is.
Especially when considering this:
* having a pull-down on GPIO 0 is not normally _not_ a good idea, as boards often have a built-in pull-up on that pin
* if GPIO 0 is low on boot, the board goes into firmware download mode
* there is no "button timer" in WLED. So its not so simple to reset a timer in the software, which doesn't even exist.

flux242, 2024-07-19:
really? Did I ever mentioned GPIO0? And no timer exists they say huh? What's that then
```

      if (b == 0 && dur > WLED_LONG_AP) { // long press on button 0 (when released)
        if (dur > WLED_LONG_FACTORY_RESET) { // factory reset if pressed > 10 seconds
          WLED_FS.format();
          #ifdef WLED_ADD_EEPROM_SUPPORT
          clearEEPROM();
          #endif
          doReboot = true;
        } else {
          WLED::instance().initAP(true);
        }
      } else if (!buttonLongPressed[b]) { //short press
        //NOTE: this interferes with double click handling in usermods so usermod needs to implement full button handling
        if (b != 1 && !macroDoublePress[b]) { //don't wait for double press on buttons without a default action if no double press macro set
          shortPressAction(b);
        } else { //double press if less than 350 ms between current press and previous short press release (buttonWaitTime!=0)
```

flux242, 2024-07-19:
if you still don't get it, I'll reiterate:
Having a function of erasing EEPROM on a button is dangerous because it happens without any warning! Resetting AP pass is perfectly fine and I like it. That's why I don't want to use Button1 which lacks this functionality.
Even if there's a timer reset for a button before applying new settings, the buttons itself could stuck because of some mechanical issue or because of overheating.

softhack007, 2024-07-19:
> if you still don't get it, I'll reiterate:

I don't need a teacher. And I don't need some dabbler who beliefs he can read the source code to me.
PERIOD.

flux242, 2024-07-19:
> > if you still don't get it, I'll reiterate:
> 
> I don't need a teacher. And I don't need some dabbler who beliefs he can read the source code to me. PERIOD.

I wasn't even referencing you with that one because it would make no sense

blazoncek, 2024-07-19:
I do not get why are you so fussed about a clearly described button 0 functionality.

> Note: Button 0 has two, built-in functions. 1. Hold it down for >6 seconds and the Wi-Fi settings will be reset to default. 2. Hold it down for >12 seconds and flash memory is erased.

Taken from [KB](https://kno.wled.ge).

If you've *soldered* your button to a default GPIO (0, which is also a strapping pin) prior to flashing you wouldn't have lost anything as you wouldn't be able to save anything.

If you did it afterwards (or changed GPIO) you were notified about its functionality in the KB.

RTFM as they say.

flux242, 2024-07-20:
its amazing you still not getting what i'm talking about but insist that i wrongly soldered something to something! I care less too, because i simply removed that erase timer completely locally and i'm fine. RTFM huh? Keep telling this to everyone for better user experience

flux242, 2024-08-14:
buttons interface is fubar. First of all they attached some dangerous effects to button0 without mentioning that in the gui! Now they say use other buttons but that isn't working either because on all buttons except button0 action is repeated automatically for long presses. So if I would want to cycle through presets with short press and switch on/off with long one it wouldn't work. And the only way to get it working is to remove all hidden actions from button0 and then use it

softhack007, 2024-08-14:
> buttons interface is fubar. First of all they [...]

Who is "they" ? As my grandpa used to say - "they" are selling cucumbers on the market 😉 .

🆒  cool down ... no need to get emotional. This is a community project, not a product made by a professional company. So no need to fight for your right.

flux242, 2024-08-14:
those who don't even admit that there's a problem with buttons

blazoncek, 2024-08-14:
> those who don't even admit that there's a problem with buttons

There isn't. 🤣 
That is a core functionality built into WLED since 0.12.

softhack007, 2024-08-15:
The behaviour is explained already in our KB
https://kno.wled.ge/features/macros/#buttons

> Note: Button 0 has two, built-in functions. 1. Hold it down for >6 seconds and the Wi-Fi settings will be reset to default. 2. Hold it down for >12 seconds and flash memory is erased.

akluchnikov, 2024-08-15:
thanks for the answers!
Yes, I already realized that this is not a bug, but a feature. But for some reason I have never encountered this, although I have already assembled 20+ devices using the same scheme...

closed 2024-12-13 by DedeHai