wled/WLED #4067: Adding virtual LED outputs (DDP) breaks original LED output config that uses a local GPIO pin

automatedelectrics, 2024-07-21:
### What happened?

* I have a WLED controller that uses a local GPIO pin to run a string of SK6812 LED's; everything is working perfectly well (v0.15.0-b4).
* I wanted to add a second controller's LED's (set up exactly the same as the first one and also running v0.15.0-b4) as a virtual output of the first controller.
* When I modify the configuration of the first controller (LED Preferences > Hardware setup > LED outputs) to add a second output as either 'DDP RGB (network)', or 'DDP RGBW (network)', and save the updated config, the WLED controller no longer works and does not run any effects.
* When I go back into the the LED output configuration settings, the settings for the second output are still there and OK, but the original settings for output 1 that use the local GPIO pin are wiped blank and I cannot reassign them as the first drop down selection box list to set the type of LED's is also blank.
* Rolling back to version v0.15.0-b3 on the first controller solves this issue, and I am able to add numerous virtual LED's to several different WLED controllers as needed.

### To Reproduce Bug

* Set up a WLED controller with v0.15.0-b4 firmware and output 1 to use a local GPIO pin to control a LED strip.
* Add a second output as either a 'DDP RGB (network)', or 'DDP RGBW (network)' type and save the updated config.

### Expected Behavior

* The WLED controller can run normally with output 1 connected to a LED strip via a local GPIO pin and output 2 connected to a virtual LED strip over an Ethernet network using DDP.

### Install Method

Binary from WLED.me

### What version of WLED?

v0.15.0-b4

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-07-22:
b4 is known to have issues in LED configuration. Most have been fixed since.
Please build from source.

automatedelectrics, 2024-07-22:
Thanks for letting me know. I appreciate the update :)

closed 2024-11-09 by github-actions[bot]