wled/WLED #4070: Pins reserved in PinManager not reflected in LED settings UI page

axlan, 2024-07-23:
### What happened?

When writing a UserMod that makes use of pins beside the global SPI/I2C, you can allocate them in the PinManager.

These pins show up as unused in the "Button N GPIO:" dropdown. Trying to save them will be rejected on the device, but their won't be any indication in the UI aside from the button value going back to disabled. 

### To Reproduce Bug

Write a usermod that allocates pins. Try settings those pins to push buttons in the web UI.

### Expected Behavior

These pins should probably not be listed as options, or disabled and indicated that they're used by a UserMod (similar to how they currently indicate they're used by other buttons/LED).

### Install Method

Self-Compiled

### What version of WLED?

v0.14.4-6-g3615ab53

### Which microcontroller/board are you seeing the problem on?

ESP32-S3

### Anything else?

The simplest fix appears to be including pins allocated by things other then the buttons/LED in the `um_p` field updated in  wled00/xml.cpp, `appendGPIOinfo()`. It's currently just including the  i2c_sda/i2c_scl/spi_mosi/spi_sclk .

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

axlan, 2024-07-23:
I made a simple fix https://github.com/Aircoookie/WLED/pull/4071

blazoncek, 2024-07-24:
This is considered a bug in usermod, not WLED.
See my comment in #4071.

closed 2024-07-25 by axlan