wled/WLED #4075: CCT with three different LEDs / PWM channels

Stefan-Olt, 2024-07-26:
**Is your feature request related to a problem? Please describe.**
Mixing two different whites will result in a color that is on a straight line between the two whites in the CIE color space. Unfortunately the white is not on a straight line, but on a curve in the CIE color space. While for higher color temperatures a line is a good approximation, for low temperatures it's not. I also tested this  and found that the light from a very warm LED (1800K) and a cold LED (5700K) mixed do not result in a nice standard warm-white.
But I would like to have a controllable white from very warm (1800K) to daylight (5700K). So I made LED boards with 3 different LEDs: 1800K, 2700K and 5700K.

**Describe the solution you'd like**
I had quick look at the source, there is only `TYPE_ANALOG_2CH` for CCT and `TYPE_ANALOG_3CH` for RGB. A new type like `TYPE_ANALOG_CCT_3CH` would be needed and a calculation for that would have to be added to `BusPwm::setPixelColor`. That seems to be the easy part, because if you want to do it properly you would not want to hard-code the color temperatures of the used LEDs in that function, but rather have it user-defined. It seems that there is some code in WLED to compensate RGB channels with white, but I couldn't find any place that actually defines the white color temperatures of the white LEDs or the wavelengths of the RGB components to do a proper calculation, is that correct or am I missing something? WW is assumed to be 1900K and CW is just white?

**Describe alternatives you've considered**
I did consider a user-mod, but that does not seem possible.

Stefan-Olt, 2024-07-26:
> Can you provide a reference to CIE curve for white?

You find the curve for example [here](https://upload.wikimedia.org/wikipedia/commons/5/5f/CIE-1931_diagram_in_LAB_space.svg). I have plotted two graphs (I used the Python colour package), both show a white curve from 1800K to 6000K (dots equally spaced 50 K apart), the other line is in one case the mix between 1800K and 5700K and in the other 2700K and 5700K (in 100 steps). As you can tell, the linear approximation is good for 2700K to 5700K, but won't work with 1800K and 5700K:

[image]
[image]

> Apart from that, using 3 whites with different color temperatures will (most likely) not bi implemented (as it would require rebuilding the CCT handling from ground up).
As of current implementation WLED has no notion about actual color temperatures of used LEDs. It just linearly distributes brightness according to CCT slider position (i.e. 0 == warm only, 255 == cool only).

Can you explain that? What else would have to be done than adding a new LED type like `TYPE_ANALOG_CCT_3CH` and add functionality in `BusPwm::setPixelColor` to set the 3 values correct? If you don't care about exact colors and just use warm white and cold white, all you would have to do is to split the range, CCT == 0 warm only, CCT == 128 middle only, CCT == 2 cool only?

blazoncek, 2024-07-26:
You are welcome to explore 3 LED CCT handling.

My only advice: Keep in mind that not everyone will have 1800K, 2700K and 5K00K LEDs and that WLED does not know anything about color temperature of white LEDs.