wled/WLED #4076: ESP32 C2 touch threshold has no effect

RobinMeis, 2024-07-26:
### What happened?

I'm running WLED 0.15.0-b4 on ESP32 S2 Mini. Adjusting the touch threshold doesn't change anything. Sensitivity doesn't change, it doesn't make any difference if I set the threshold to 1 or to 100. Sensitivity is always too low.

### To Reproduce Bug

Attach a touch surface to the board and play around with the touch threshold.

### Expected Behavior

Changing the threshold should change the sensitivity

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0-b4

### Which microcontroller/board are you seeing the problem on?

ESP32-S2

### Anything else?

I think this is related to https://github.com/Aircoookie/WLED/pull/3798 and the shifting of touchThreshold in https://github.com/DedeHai/WLED/blob/15526bd6e8c7a75ee1e36cf24ae397a674f63c56/wled00/cfg.cpp#L260

`touchAttachInterrupt(btnPin[s], touchButtonISR, touchThreshold<<4); //threshold on Touch V2 is much higher (1500 is a value given by Espressif example)`

Maybe the shifting causes a scaling issue.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-07-26:
This change was implemented by @dedehai and will require his attention.

RobinMeis, 2024-07-26:
I started to compile the firmware on my own and played around with the mentioned threshold scaling. I could achieve the desired sensitivity by lowering the threshold significantly. In my tests including touch buttons behind plastic covers, values around 100 worked best.

With ESP32 Wroom I observed that I had to increase the threshold to get a higher sensitivity but for Touch V2 it seems to be opposite. Setting the threshold < 100 results in ghost touches for some buttons. On the other hand setting the threshold static to 65300 results in no sensitivity at all.

So I think that we need to find a proper range to implement a better scaling. I modified the scaling to allow values between 50 and 3050. At least for the buttons I tested with, the range seems to be suitable.

I pushed my current changes to https://github.com/RobinMeis/WLED/blob/9c7afedece573d0611541cfe372f48fbd9b0c056/wled00/cfg.cpp#L299

RobinMeis, 2024-07-26:
>  What was the exact geometry and setup of your touch button?

For testing I used a round copper plate with a diameter of 18mm behind several thin plastic cards. Without the plastic cards it worked. The real "button" is actually a coil spring that is part of a light.

Both worked well with ESP32 Wroom, but I could only get them working with ESP32 S2 after the described modifications.

DedeHai, 2024-07-26:
> So I think that we need to find a proper range to implement a better scaling. I modified the scaling to allow values between 50 and 3050. At least for the buttons I tested with, the range seems to be suitable.

Bit contradictory. The code uses a range from 0 to 4080 (at least it should, no PC access RN), which you say does not do anything. Could it be just a missing typecast?

RobinMeis, 2024-07-26:
It doesn't. What I replaced from current 0_15 branch was
`256 + (touchThreshold << 8)`
which won't allow to go below 256

GHBLoos, 2024-09-07:
I have the same problem. I like to check my config first to exclude a possible wrong setting or so, before I build and flash. 

**ESP32 wemos S2 mini
version v0.14.4 "Hoshi"**

In prefeferences:
**Button 0 GPIO 14 - Touch**
Currently I have a 10cm wire connected to GPIO 14 with a stripped end of about 4 cm

@RobinMeis : Do you have a comparable config?

DedeHai, 2024-09-07:
V0.14 has no touch support for S2/S3, that bug is fixed in latest 0.15

cameronnathan707, 2024-09-22:
Hi, I have not been able to get touch working with a 5cm wire using the "SEEEDSTUDIO XIAO ESP32S3". I have tried on multiple pins and on multiple boards with the version 0.15.0-b5.

DedeHai, 2024-09-22:
If you need help with your setup, this is not the place, please use Discord or Discourse forum. Touch works fine on S3 in 0.15-B5.

closed 2024-10-27 by blazoncek