wled/WLED #4079: ESP-01 web interface not loading when connected to Wi-Fi and version >13.3

einsnulleins, 2024-07-31:
### What happened?

Figured I'd update my ESP-01S after not doing it for a long time and noticed that the webinterface won't load anymore. Takes a couple of seconds for the browser to grab the WLED title and the favicon but beyond showing the dark background color, nothing happens. (I might have to wait for several minutes for anything to happen but that can't be the intended functionality)
Tried several versions from 14.0 to the latest 15 b4; Same issue.

Before connecting it to wifi and using its own WLED-AP, the interface works fine.

The first v14 beta in /releases kinda sorta works but is VERY VERY VERY slow compared to 13.3. Just barely useable.

Verified that the flash contents match the individual wled binary.

Went back to 13.3 for testing and everything works perfectly.

### To Reproduce Bug

Update ESP-01S to a version later than 13.3, connect it to your local wifi, try to open the web interface.

Router is a AVM Fritz!Box 7590. Might be related as AVM has some weirdness going on with some devices...

### Expected Behavior

everything just works.

### Install Method

Binary from WLED.me

### What version of WLED?

anything newer than 13.3

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell
Please let me know how to grab logs. Serial output does not list anything after boot and the web interface stays blank.
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2024-07-31:
@einsnulleins  Which one is it esp01 or esp01s  . The esp01s i have seems to be working fine on 15.0.b3 but will try latest once you confirm 

[image]

To see the logs on serial you would first have to compile a build  with those flags 
build_flags = ${common.debug_flags}
monitor_filters = esp8266_exception_decoder

Might be first step is try to download the latest bin and flash it with this gui tool ( i never use WLED.me site with esp01s )   
https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/WLED_%20ESP_Flasher

einsnulleins, 2024-08-01:
@dosipod It's an ESP-01S. I used the binaries from /release but the drop down in the issue template only offered wled.me or self-compiled.

I used ESP-flasher and the official esptool. Both fine with 13.3, both borked with anything newer. The 160mhz bin also seemed to behave the same.

Got a few other things going on and I only wanted a quick and easy upgrade, will see when I find the time to build a debuggin binary. Would be very helpful if someone could provide it.

dosipod, 2024-08-01:
I have faced issues with some esp01s due to it being abused or faulty but also faced connectivity issues on old builds . I did not notice issues at the moment as you see in the pic so it could be something with your environments  . If you are facing the issues with multiple esps then might be clarify that . In all cases you can use this helper page to compile debug builds  https://wled-compile.github.io/

blazoncek, 2024-08-01:
Fritzbox is known to cause issues/problems with WLED. Search other issues opened for possible solution.

There were several other issues opened wit the same title so continue there, I'm closing this duplicate.
Please also use WLED [forum ](https://wled.discourse.group)or Discord for help and support questions.

closed 2024-08-01 by blazoncek

einsnulleins, 2024-08-01:
@dosipod It's an ESP-01S. I used the binaries from /release but the drop down in the issue template only offered wled.me or self-compiled.

I used ESP-flasher and the official esptool. Both fine with 13.3, both borked with anything newer. The 160mhz bin also seemed to behave the same.

Got a few other things going on and I only wanted a quick and easy upgrade, will see when I find the time to build a debuggin binary. Would be very helpful if someone could provide it.

--------

thanks for closing without pointing me anywhere.
I searched the github issues for several minutes and tried two dozen entries without finding anything.
And even if this company's devices are known to cause issues (hence me mentioning it) how is it that the old builds are fine?

At least give me *something* to work with here.

will scour the forum; maybe there's another user who has an idea...

and screw discord.