wled/WLED #4080: Brave Browser not recognized

RodBarnes, 2024-08-01:
**Is your feature request related to a problem? Please describe.**
Currently, the app limits to only the Chrome and Edge browser.  Why not all Chromium-based browsers?  Surely there isn't some Google- or Microsoft-specific code requirement.

**Describe the solution you'd like**
Please change it so the app will allow running on any Chromium based browser not just Chrome or Microsoft.

**Describe alternatives you've considered**
No alternative for those who refuse to let Google and Microsoft steal their data.

Thank you for your ideas for making WLED better!

blazoncek, 2024-08-01:
App? Which app? WLED is not "an app". It is firmware for ESP line of SoC.
If you are talking about WLED UI, it supports all modern browsers.
If you are talking about installer then open an issue [there](https://github.com/Aircoookie/WLED-Webinstaller) (although browser needs to support SerialAPI to successfully flash ESP which currently limits chioces to Edge or Chrome AFAIK).
If you are talking about mobile app, the official has been discontinued. The alternative is WLED Native it will use web-view provided by mobile device.

dosipod, 2024-08-01:
He means the installer(s) , which is not limited to wled and related to Web Serial API , @RodBarnes  List is here https://caniuse.com/web-serial  if you want to open an issue 
[image]

Cheers

RodBarnes, 2024-08-01:
@dosipod Thanks for the clarification and the link.

blazoncek, 2024-08-01:
Closing as not WLED firmware related.

closed 2024-08-01 by blazoncek