wled/WLED #4085: Playlist Stops Changing When Timer Set and Browser Closed

quadbock, 2024-08-03:
### What happened?

When starting a playlist, if the Timer is set and then the webpage is closed, the playlist does not continue and is stuck on the current effect. 

### To Reproduce Bug

1. Start a Playlist 
2. Click on the Timer
3. Exit the page or close the browser

### Expected Behavior

The Playlist would continue and then WLED would shut down after the designated Timer time.

### Install Method

Binary from WLED.me

### What version of WLED?

2406290

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-08-04:
You can configure playlist to stop at the pre-determined time without activating *nightlight*.

Nightlight AKA timer is incompatible with playlist.

quadbock, 2024-09-06:
Problem still exists.  I left out important info.  When I stated: "When starting a playlist, if the Timer is set and then the webpage is closed, the playlist does not continue and is stuck on the current effect."

I forgot to mention that if I did NOT close the webpage, everything worked as expected."

Thanks.