wled/WLED #4091: Allow ADC attenuation for audio input to be selected

lvml, 2024-08-04:
**Is your feature request related to a problem? Please describe.**

Documents such as [this](https://github.com/atuline/WLED/blob/assets/docs/Microphones.pdf) or [this](https://github.com/atuline/WLED/wiki/Analog-Audio-Input-Options) describe "cheap microphone modules" as unusable for the "audio-reactive" modes primarily because the range of digital values obtained from the "cheap" microphones via the ADC is small.

However, in https://github.com/Aircoookie/WLED/blob/main/usermods/audioreactive/audio_source.h#L625 a fixed attenuation of 11db is deliberately set, [in expectation of analog input between 150mV to 2450mV](https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32/api-reference/peripherals/adc.html#adc-attenuation). This, of course, reduces the range of values to expect from the analog to digital conversion from "cheap" microphone modules that happen to deliver a smaller analog voltage range.

**Describe the solution you'd like**

Make the parameter to the adc1_config_channel_atten() call at https://github.com/Aircoookie/WLED/blob/main/usermods/audioreactive/audio_source.h#L625 subject to a configuration parameter, such that no attenuation (ADC_ATTEN_DB_0) can be selected.

**Describe alternatives you've considered**

Buying "better" microphones is probably always an option, but given that this is just about some lights blinking according to the music, this should really not require "high fidelity" input.

**Additional context**

none

k7bbr, 2024-08-05:
> Analog microphones will be deprecated in 0.15 and removed in 0.16.
> Reasons being removed I2S support for analog in newer frameworks and poor
> performance.
> @softhack007 <https://github.com/softhack007> can elaborate in more
> detail.

Hope you don't mind if I chime in to say that I've never seen poor
performance with my analog mics and analog audio input from an audio
source. Would love to see analog support continue if it's possible. It's
nice to just have to use one pin for input on setups where there's other
buttons and sensors hooked up and available pins are an issue.

lvml, 2024-08-05:
> Analog microphones will be deprecated in 0.15 and removed in 0.16.
> Reasons being removed I2S support for analog in newer frameworks and poor performance.

I2S support is [still documented for the latest 5.3 Espressif IDF](https://docs.espressif.com/projects/esp-idf/en/v5.3/esp32/api-reference/peripherals/i2s.html) and the [Analog to Digital Converter (ADC) Continuous Mode Driver](https://docs.espressif.com/projects/esp-idf/en/v5.3/esp32/api-reference/peripherals/adc_continuous.html) is still documented to make use of I2S, internally.

Regarding "poor performance", I wonder what measurement this refers to.

softhack007, 2024-08-06:
> I2S support is [still documented for the latest 5.3 Espressif IDF](https://docs.espressif.com/projects/esp-idf/en/v5.3/esp32/api-reference/peripherals/i2s.html) and the [Analog to Digital Converter [...]

If you know how to make use of the continuous ADC mode driver for audio sampling, please feel free to to write a driver for WLED and submit a PR. Just be aware that the I2S driver we currently use does not support ADC input any more on newer MCU's.

> Regarding "poor performance", I wonder what measurement this refers to.

"Poor performance" is based on own experience, by comparing microphone modules like the MAX9814 (analog) with I2S mics like the IMNP441. In this comparison, the MAX loses, and the difference between the two devices is so huge it should be measured in lightyears. 

Please also keep in mind that ADC on esp32 is very poor, compared to I2S. ADC has a very noisy 12 bit resolution, while I2S is delivering 24bits per sample.

So - given that ADC is deprecated in the I2S driver, given the low quality of ADC, and given that good I2S devices are available for cheap - the WLED core team has decided to focus our limited time and effort on I2S hardware, and drop I2S-ADC in the future.