wled/WLED #4097: Home Assistant connection error after update to 14.4 stable when creating segments

Boerny41, 2024-08-10:
### What happened?

Home Assistant forced me to update WLED (13.1). After doing the update, I got an error saying:

[image]

I decided to do a factory reset. After that I was able to connect it to Home Assistant again, but once I re-did my segments (8 in total, 13 LEDs each) it broke again.

I've now reset the WLED instance 3 times, every time it seems fine, but when all segments are created it breaks. I've tried adding one by one, and it seemed to work with 5 segments but broke with 8

Home Assistant is latest stable as is WLED

### To Reproduce Bug

Add a WLED instance with 8+ segments to Home Assistant, click the refresh button or restart home assistant

### Expected Behavior

no error

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.1 stable

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Anything else?

The refresh button I keep talking about

[image]

[image]

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-08-10:
Yo may want to upgrade esp to ESP32 as many segments and HA are not best friends.
0.14 (including 0.14.4 use more RAM) than previous versions.

or reduce number of segments.

Boerny41, 2024-08-10:
I'm not sure if that's the problem. I've just added 10 segments to the other device you could see on the screenshots. It now got 11 segments, 190 LEDs and 72 entities in HA. Same patter and color and both saved as preset but no problems there even though it is the same but more in every way.

[image]

blazoncek, 2024-08-10:
I am not familiar with HA so you may be better off opening an issue on HA repository as the error messages are HA's not WLED's.
Be sure you use the most recent version of WLED (0.14.4).

Plawasan, 2024-08-20:
https://github.com/home-assistant/core/issues/123506 fixed this for me in HA 2024.8.2

Zerobyte1, 2024-09-16:
You do not need to reset wled. if you goto http://<wled address>/edit you can just delete the faulting preset file and then HA will be fine.

closed 2025-01-21 by github-actions[bot]