wled/WLED #4100: ESP32-S3 DevkitC N16R8 not flashing and not working

MangyCat, 2024-08-13:
### What happened?

So i've been trying to install WLED on my esp32-s3 devkit so i can connect some leds to it and glue it to my tv  for the end of eternity, but while flashing the board alot of errors and disappointments arose, like i tried the USB C port with native serial from the chipset, and tried the built in devboard serial chip usb side, and the only thing that happened was bootloops which i knew from the TX led blinking fast or a error message. The board works when flashing arduino examples (latest IDF) but doesnt when flashing WLED, i tried esptool.py, the old installation method, the github.io method, the official website method, nothing would work and i tried every version.

### To Reproduce Bug

Install WLED through every installation method imaginable
Wait for it to install
Reset device
be disappointed when the TX led is flashing and entering into a serial is full of bootlooping and even crashes your terminal

### Expected Behavior

I expected the WLED menu to pop up and finally to connect my WS2812B strips and configure it as usual

### Install Method

Binary from WLED.me

### What version of WLED?

14.4 attempted beta 15 attempted

### Which microcontroller/board are you seeing the problem on?

ESP32-S3

### Relevant log/trace output

```shell
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2920
entry 0x403c98d8
E (261) esp_core_dump_flash: No core dump partition found!
E (261) esp_core_dump_flash: No core dump partition found!
```

### Anything else?

this is where i got the board from https://www.aliexpress.com/item/1005005051294262.html

also my ESP32-S2 wemos mini board works just fine with WLED, why couldnt this board just work, i bought the board as an replacement to my trusty ESP32 that died a tragic death of reverse polarity

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-08-13:
This may be an issue regarding correct flash/PSRAM mode and not WLED code.
There are QSPI and OPI variants and, as learned recently, OPI/OPI variant.
You may need to experiment which binary to flash by custom compiling and flashing using PlatformIO.

Once you figure out correct options open an issue on WLED installer Github repository.

Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.

closed 2024-08-13 by blazoncek

3xtant, 2024-08-19:
I agree with @blazoncek about this being a matter of choosing the correct variants, rather than a bug. I had the same issue with an ESP32-S3-DevKitC N8R2. I did what probably most people do looking for answers everywhere, but there are few good answers and most are outdated since support for these boards is relatively new (as of 2024).

In case this helps others, this what worked for me:
I looked at espressif's idf tool, which I installed following the VS Code extension method.
Found my board and built and uploaded a couple of their examples to make sure it worked as expected.
I familiarized myself with the memory configuration and options.
Then, I forked and cloned WLED and followed the PlatformIO instructions.
Once PlatformIO was configured, I edited `platformio.ini`. Turns out I only had to enable an existing environment:
```
default_envs = esp32s3dev_8MB_PSRAM_qspi
```
After that single change, my board built, uploaded and started working right away.

I suspect the N16R8 board is very similar and might need tweaking of the environment above. In that case you would create a new file in the same folder as `platformio.ini` called `platformio_override.ini`. You would add extra items or tweak existing ones there. Note that there is an `esp32s3dev_16MB_PSRAM_opi` environment, but that will only work if your memory is octal rather than quad. That's why I mentioned the bit about downloading the idf tool and testing your board. That should help you determine the proper settings from the espressif files.